slic3r/Slic3r #2996: Manual refers to non-existant menu entry

PxT, 2015-07-14:
In the multi-material section, the manual refers to a "Combine multi-material STL files..." menu option. I couldn't find this in the latest v1.2.9. I figured out the proper procedure, but it would be nice to have the manual updated.

http://manual.slic3r.org/expert-mode/multiple-extruders

EricSteijlen, 2015-07-17:
Hi PxT,
Can you tell me what the 'proper procedure' is please?

xoan, 2015-07-17:
There is not so much activity in a https://github.com/alexrj/Slic3r-Manual/issues, but you can try to issue this there or fix it in your own fork (it is LaTeX) and make a pull-request.

PxT, 2015-07-17:
Moved to https://github.com/alexrj/Slic3r-Manual/issues/28

@EricSteijlen Load the first part like normal, then select it and choose *Settings* from the icon bar across the top, then the *Load Part* button to add additional parts.

closed 2015-07-17 by PxT

smartroad, 2015-07-17:
Awesome, have just got my dual extruder working and was wondering how to make this work as just 'Add...' lost the Z offset height :+1: