slic3r/Slic3r #2999: Wx install fails because of missing Carbon.r

MartinMajewski, 2015-07-18:
I reinstalled Yosemite and wanted to build slic3r. The non-gui build went through well, but the gui build is still a pain in the a**.

The main problem is, that wx won't install no matter what. Looking through the log I noticed, that it somehow depends on a "Carbon.r" file.

```
7 warnings generated.
echo '#include <Carbon.r>' > cpp/wxPerl.osx/wxPerl.r
cd cpp/wxPerl.osx && make RESTOOLDIR='/Applications/Xcode.app/Contents/Developer/Tools' ARCH_FLAGS=''
mkdir -p build/wxPerl.app/Contents
cp Info.plist build/wxPerl.app/Contents/Info.plist
mkdir -p build/wxPerl.app/Contents/MacOS
gcc -o build/wxPerl.app/Contents/MacOS/wxPerl  main.c
mkdir -p build/wxPerl.app/Contents/Resources
/Applications/Xcode.app/Contents/Developer/Tools/Rez -d __DARWIN__ -useDF -o build/wxPerl.rsrc  wxPerl.r
failed to find Carbon/Carbon.r
wxPerl.r:1: ### /Applications/Xcode.app/Contents/Developer/Tools/Rez - SysError 0 during open of "Carbon.r".
Fatal Error!
wxPerl.r:1: ### /Applications/Xcode.app/Contents/Developer/Tools/Rez - Fatal Error, can't recover.
Carbon.r: ### /Applications/Xcode.app/Contents/Developer/Tools/Rez - Since errors occurred, build/wxPerl.rsrc's resource fork was not written.
make[1]: *** [build/wxPerl.app/Contents/Resources/wxPerl.rsrc] Error 3
make: *** [wxPerl] Error 2
  MDOOTSON/Wx-0.9927.tar.gz
  /usr/bin/make -- NOT OK
'YAML' not installed, will not store persistent state
Running make test
  Can't test without successful make
Running make install
  Make had returned bad status, install seems impossible
```

Has someone any idea what this is, where to find it and how to finally get Wx running? Alien::wxWidgets is installed, but that doesn't help at all. :-(

alf239, 2015-12-16:
Hi Martin, I'm trying to solve a similar problem with wxHaskell at the moment, and found a wiki which may be relevant, check it out: http://wiki.bolay.net/doku.php?id=acdsn:acdsn-a:mac

> Find your `Carbon.r`.
> 
>     $ sudo find / -name Carbon.r -print
>     $ # => /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/Carbon.framework/Versions/A/Headers/Carbon.r

> We're only interested by the path up to the Frameworks directory.

> Go to your Wx `build` directory.

>     $ cd ~/.cpanm/work/timestamp.something/Wx-0.9927

> Edit `cpp/wxPerl.osx/Makefile` and add the path to the `Frameworks` to the `Rez` call (line 36).

Not sure whether it helps, but it looks promising.

alexrj, 2017-04-03:
Thank you for reporting this. The instructions are https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X are tested on MacOS X 10.12 Sierra. I patched the Alien::wxWidgets and Wx modules to work on newer OSX since the maintainer is inactive.

closed 2017-04-03 by alexrj