slic3r/Slic3r #3000: Dual Extruder: Which extruder is starting?

bonm014, 2015-07-18:
I have a dual extruder setup and want only to start heating the extruder that is needed first (Could be 0 or 1, depends on the object). At the tool change gcode I will heat up the other one.
Is there a possibility to have a [next_extruder] inside the start gcode? The manual does not describe this...
Also it would be nice if [first_layer_temperature_next] is also available within the startcode so the extruder that will be used (next_extruder) will get the correct temperatur (first_layer_temperature_next) 

This way I do not have to start heating both extruder and worry about oozing on my print.

Or just kickof the toolchange gcode after the start code, this will eliminate my problem aswell.

Slic3r: 1.2.9
Operating System: Windows 8.1
Steps to reproduce:
- Create a printer setting with dual extruder and create a start gcode with [next_extruder] for the M109 SS[first_layer_temperature_0] T[next_extruder] line

Patola, 2015-08-23:
I also need that feature (more variables to use in the custom gcode) and as soon as it is implemented I'll be eager to test.

markwal, 2015-08-24:
1.2.9 has [current_extruder]. It was added for a similar problem on layer change for people that wanted a purge tower per extruder #2866 . However, I'm not sure it will help us in the start gcode or not.  It seems to be defined, but in my brief testing it seemed to be set to 1 for my start gcode but then shortly after it switched to 0 (and called the layer change before and after gcode but not the tool change gcode) before the first extrusion.  Which was weird.

I tried this in my start gcode: `M104 S[first_layer_temperature_[current_extruder]] T[current_extruder]` and it got translated to `M104 230 T1`, but then it emitted the following immediately after the start gcode (I had comments turned on):

```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
; before layer change
G92 E0 ; reset extrusion distance
T0 ; change extruder
G92 E0 ; reset extrusion distance
; after layer change
```

maukcc, 2016-11-15:
I love this [current_extruder] placeholder.
as markwal says it is not perfect:

When using a single part object and selecting either extruder 0 or 1 to make it with, it will work OK.

When using a dual part object it will start printing with the part that is being done with extruder 0, but [current_extruder] is populated with the first part (0 in the screenshot)

[image]

maukcc, 2016-11-15:
lets further explain:

[image]

The base part here are the Cartesio letters (dark green) that will be done with extruder_0
The ground plane part (red) should be done with extruder_1
But because all ground plane parts start with extruder_0 it will not be correct until a tool change is being done

maukcc, 2016-11-15:
2 errors here:
Start printing with the extruder that was assigned to the part on the ground plane
Populate the [current_extruder] placeholder with that extruder value

maukcc, 2016-11-17:
Sorry, this is not correct, it starts with the correct extruder, but I had a brim that was set to be made with extruder_0. So that was why it always started with extruder_0

maukcc, 2016-11-23:
And that was also not correct :)
Brim is always made with the extruder assigned to the 1st part loaded
Skirt is always made with the extruder with the lowest value