slic3r/Slic3r #3001: Raft pattern angles

wfcook, 2015-07-20:
I've noticed a couple of things about rafts that are less than ideal:

(1) The first two layers of the raft are printed at the same pattern angle, which results in the first layer (thicker extrusion) not properly supporting the second layer (thinner).

(2) The first interface layer is printed at the same pattern angle as the first raft angle, no matter what angle the prior raft layer was printed at. Again, this results in improperly supported interface material unless an odd number of non-interface raft layers is used.

(3) All interface layers are printed at the same angle. This is not necessarily a problem, but not what I expected and I would think it would be better if they alternated like the other layers. What it the reason for this?

Thanks,
Bill Cook _/)_

GregoryNoel, 2015-07-24:
I experienced a problem with interface layers in the same direction. The first layer was not perfectly sticking to the support material (5mm height) and then the second interface layer was imperfect.
The first layer above the support material is in fact very poor in quality because not sticking well on the irregular height of interface layer -> Very poor finish on this object layer.

GregoryNoel, 2015-08-23:
I'm not expert in perl, but I've tried changing angles. I add 2 lines in SupportMaterial.pm.
Interface layers are now each time rotated.

If someone want to try, to enhance and integrate it, feel free.

[image]
[image]
[image]
[image]

a4jp-com, 2015-08-24:
I can't understand what is happening here. My layers are working and each layer is rotated 90 degrees from the last so everything seems to work. How do you make the supports? Do you have a simple test file? Maybe a cube with supports on the side or something.

[image]
[image]
[image]

wfcook, 2015-08-24:
The issue is only with rafts and interface layers

a4jp-com, 2015-08-25:
If I go to printer settings > support material > raft layers and type in a value like 4 I can see the layers don't get rotated as well. Can you fix this @alexrj ?

I'll include some pictures below.
-------------------------

I guess the first layer is being smooshed down so the lines are bigger. This can probably be fixed in the printer settings > advance settings area. Make the first layer the same size as the other layers to fix the problem. After that you couls try adjusting the z-depth in printer settings > size and coordinates > z offset

The other problem of spacing and heat/surface bubbling are probably fixable but you need to upload your config file to get help there. File > export config.

--------------

[image]
[image]
[image]
[image]
[image]
[image]

koghe, 2015-08-30:
hi i asked almost the same question to alexrj some time ago, raft rise the object and it add supports
(https://github.com/alexrj/Slic3r/issues/1323#issuecomment-21620753 )

+++1 to have a alternate angle for raft like the old skeinforge

dn63, 2018-03-05:
Opened pull request #4334 with code changes suggested by @GregoryNoel (with minor edits).

lordofhyphens, 2018-04-05:
I've accepted the PR; we'll see how it behaves (and if anyone complains about the behavior :) )

Thank you @dn63

commit 6454abfd (2018-04-05): Raft pattern angles (#3001) (#4334)

Fixes #3001
closed 2018-04-05 by lordofhyphens