slic3r/Slic3r #3003: Bad STL import

wfcook, 2015-07-20:
I have an STL file which loads fine in Cura, Makerware, Simplify3D, and a bunch of different CAD packages. Slic3r seems to blow up on it. Is this something the developer would like to look at, and if so how do I send it to him?

kefir-, 2015-07-20:
Share it from Dropbox or Google drive, or upload it on the reprap forum, or somewhere else you can share a link to.

lordofhyphens, 2015-07-20:
You should also post the version you are using and a copy of your slic3r
configuration.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

wfcook, 2015-07-21:
Here's a link to the STL file:

http://www.wfcook.com/Slic3rTest.zip

wfcook, 2015-07-21:
I am using version 1.2.9, and the problem  seems independent of configuration.

wfcook, 2015-07-23:
How do I remove the "Needs STL test file" label? Or is that for alexrj to do?

lordofhyphens, 2015-07-23:
That is an @alexrj thing.

luisonoff, 2015-07-26:
I have tested it and I confirm that it works fine on Cura, but not on slic3r 1.2.9.
Slic3r says it is broken, tries to autofix it, but model breaks and misses great part of the object (on 3D view and also on preview).
I did try to fix model on Fusion360, but it says that mesh is too dense with more than 80k facets. In my opinion model does not look so complex to have so many facets...  this could be somehow related?

Another example of "bad STL import" is this print http://www.thingiverse.com/thing:745523/#files, version 2. In this case Slic3r does not complain about broken model, 3D view looks perfect but preview is broken with an extra top layer that was not supposed to be. Cura worked fine.
This time I got to fix it using Fusion360, just converting it to a solid model and exporting again.

My little conclusion: this errors are caused by bad models, but it is true that Slic3r could fix/improve this a bit, because they work on Cura.