slic3r/Slic3r #3005: The start and end codes

pOmelchenko, 2015-07-21:
I would like to be able to change the entire block start and end codes, about the same as it is implemented in kisslayser.

Because now I can not without manual correction gkoda get started automatically when the power supply to send the code to print. As seen in the screenshot what I enter is inserted into the middle of the starting block.

[image]

I would also like to have the opportunity to ask the math in these blocks. For example parking at the end of printing 15 layers from the last

G0 X20 Y155 Z_lastLayer+15_ F15000; Present Part

such a solution is implemented in the same kisslayser

alexrj, 2015-12-19:
Please read the tooltip for the Start G-code option. You'll learn how to reposition the temperature commands.

Regarding math, you don't need math for that.
Just use relative mode:

```
M83 ; use relative mode
G1 Z15 F15000
M82 ; use absolute mode
G1 X20 Y155
```

closed 2015-12-19 by alexrj

Mikk36, 2015-12-19:
M82 and M83 are wrong ones to use in this case.
Should use G90 and G91 instead.

alexrj, 2015-12-19:
Oh you're right. I fixed my example above now.