slic3r/Slic3r #3006: first .STL loads in LAYER mode only other mode Blow up

rodboudreaux, 2015-07-21:
I have been using slic3r since v1. Thanks for your fine product.

Problem: Slic3r 1.2.9 blows up when first .stl is loaded ( at end of load ), if 3D, 2D or Preview mode was selected.  If .stl is loaded in Layers mode it loads correctly and works well in all ways. Once started successfully and something has been loaded in Layers mode it will then load .STLs and function excellently in any mode.

system: Asus  intel quad 3.0 running Ubuntu 14.10

Please advise

Rod

lordofhyphens, 2015-07-21:
Please read the contributing guidelines and provide the additional
information listed there.

Additionally, I recommend using either the ubuntu package from hyperair's
repository or building your own from mine.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-12-19:
@rodboudreaux, any chance to get the required information? Otherwise no action can be taken on this report.