slic3r/Slic3r #3008: Flow error on small details

luisonoff, 2015-07-21:
Looks to me like a flow bug or even a layering bug, but maybe I am setting something wrong.

I am printing this test: http://www.thingiverse.com/thing:916214
My extruder steps are calibrated and just checked twice.
I have always used Cura but wanted to switch to new Slic3r. Installed Slic3r 1.2.9 on Windows 8.1 64 bits.

A hint: at first I thoutgh it could be a flow calculation error, mainly because with same settings and same model, Cura calculates 2.486cm3 and Slic3r says 2.7 cm3. But took a closer look to layer view and it looks like at those layers perimeters are too close form each other, while my extruding width is constant (or I set so) during whole print.

Any comment is much appreciated. Thank you

## Attempt 1
First I set everything to test autospeed, this config:
# generated by Slic3r 1.2.9 on Sat Jul 18 20:53:13 2015
avoid_crossing_perimeters = 1
bed_shape = 84.5344x8.88492,83.1425x17.6725,80.8398x26.2664,77.6514x34.5726,73.6122x42.5,68.7664x49.9617,63.1673x56.8761,56.8761x63.1673,49.9617x68.7664,42.5x73.6122,34.5726x77.6514,26.2664x80.8398,17.6725x83.1425,8.88492x84.5344,0x85,-8.88492x84.5344,-17.6725x83.1425,-26.2664x80.8398,-34.5726x77.6514,-42.5x73.6122,-49.9617x68.7664,-56.8761x63.1673,-63.1673x56.8761,-68.7664x49.9617,-73.6122x42.5,-77.6514x34.5726,-80.8398x26.2664,-83.1425x17.6725,-84.5344x8.88492,-85x0,-84.5344x-8.88492,-83.1425x-17.6725,-80.8398x-26.2664,-77.6514x-34.5726,-73.6122x-42.5,-68.7664x-49.9617,-63.1673x-56.8761,-56.8761x-63.1673,-49.9617x-68.7664,-42.5x-73.6122,-34.5726x-77.6514,-26.2664x-80.8398,-17.6725x-83.1425,-8.88492x-84.5344,0x-85,8.88492x-84.5344,17.6725x-83.1425,26.2664x-80.8398,34.5726x-77.6514,42.5x-73.6122,49.9617x-68.7664,56.8761x-63.1673,63.1673x-56.8761,68.7664x-49.9617,73.6122x-42.5,77.6514x-34.5726,80.8398x-26.2664,83.1425x-17.6725,84.5344x-8.88492,85x0
bed_temperature = 0
before_layer_gcode = ;LAYER:[layer_num]
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 5
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = \nM25 ; Pause SD print
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 0
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 10
filament_colour = #8000FF
filament_diameter = 1.75
fill_angle = 45
fill_density = 20%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 0
first_layer_extrusion_width = 250%
first_layer_height = 0.3
first_layer_speed = 30
first_layer_temperature = 0
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 0
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 25
[... 200 lines cut ...]
top_infill_extrusion_width = 0
top_solid_infill_speed = 20
top_solid_layers = 3
travel_speed = 120
use_firmware_retraction = 1
use_relative_e_distances = 0
use_volumetric_e = 1
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0

Got this print, much better, but still top layers of cube and cilinder look very ugly, it even looks like some infill was put on last layer but no perimeter.

[image]

## Cura test with same settings
I know comparing slicers is not helpful, but just wanted to show that my printer is not (I believe) the problem.

[image]

## Another example
I tried a different print, a very simple hollow cube, columns show same flow problems. Dont have this print configuration file. (in picture cube is rotated with columns facing up)

[image]

a4jp-com, 2015-07-21:
Can you print the hollow cube perfectly with Cura?

luisonoff, 2015-07-21:
Yes. This was a few days ago and I gave it to somebody as a present, but it looked perfect.
I will print it tomorrow again to upload picture.

teppix, 2015-07-22:
This looks like a duplicate of https://github.com/alexrj/Slic3r/issues/2960

luisonoff, 2015-07-22:
Thank you teppix. It was an interesting reading. Every day learning something new. :)

I am 99% sure that my attempt numer 1 and the hollow cube are suffering from the gap filling bug on #2960 as you said.

Still, at my attempt number 2 gap filling was disabled and still got flow troubles on top surface of small details. Any idea?

I will test Slic3r 1.2.7 beta and also to increase extruding width to 0.45mm.

luisonoff, 2015-07-22:
I went further looking with deatil at layer view, I posted my findings on another issue #3010 as I thougth they could be useful.

teppix, 2015-07-23:
@luisonoff 
Yes, that looks like a separate problem. I haven't noticed any such flow rate calculation problems, other than in the gap fill. Problems like that can occur if you have a high flow rate / temperature ratio though. This causes the pressure to build up in the extruder, and at the layer change, the excess plastic in the nozzle will be released, so for each layer there will be a buildup of plastic that accumulates over the height. This can also be part of the cause for the strings that you see in your second print.

I always used to print at around 190-195 C, but with recent releases I had to go up to 210-220 when there are wide extrusions in the print because of this.

I can't tell for sure from your settings if this may be the problem here. What temperature do you print at? The settings you posted says 0.

I haven't really studied your settings in detail, but I see that you decreased the speed (max speed) in the second print. Make sure that you really print at very low speeds to see if this helps.

I would also decrease the extrusion multiplier by a slight amount. Just enough to loose the ridges on the top layers. This is probably not enough to cause the problems on the small parts though, but it could help a bit.

luisonoff, 2015-07-23:
Thanks a lot for your suggestions.
That print (attempt 2) was made with ABS at 250ºC (setting stored in printer), at very low constant speed of 20mm/s, layer heigth 0.2mm. So I believe my flow is rather low and my temperature rather high.
I will try lowering the extrusion rate 1% to see if it changes, but again, I believe my extruder steps are well calibrated.

luisonoff, 2015-07-31:
I think I have found why my attempt number 2 and the hollow cube where coming ugly even with gap filling disabled. I was using firmware retractions and there was a "bug" in my printer firmware related to firmware retractions, it was extruding 1 cubic mm extra filament after each retraction...  

It is not a slic3r bug. And the gap filling bug of attemp 1 is already described in other issues, so I am closing this issue. Sorry for bothering.

closed 2015-07-31 by luisonoff