slic3r/Slic3r #3009: X/Y differences when "Detect thin walls" is disabled part duex

Praesmeodymium, 2015-07-22:
Slic3r version 1.2.9
Win 7 ultimate 64bit

When slicing this model
http://www.thingiverse.com/thing:550604/#files vase.stl

I see differences in the gcode output in the XY output near the neck depending on whether thinwall detect is on or off.

[image]
[image]

config follows

 # generated by Slic3r 1.2.9 on Tue Jul 21 20:16:30 2015
avoid_crossing_perimeters = 0
bed_shape = 79.5618x8.36228,78.2518x16.6329,76.0845x24.7214,73.0836x32.5389,69.282x40,64.7214x47.0228,59.4516x53.5304,53.5304x59.4516,47.0228x64.7214,40x69.282,32.5389x73.0836,24.7214x76.0845,16.6329x78.2518,8.36228x79.5618,0x80,-8.36228x79.5618,-16.6329x78.2518,-24.7214x76.0845,-32.5389x73.0836,-40x69.282,-47.0228x64.7214,-53.5304x59.4516,-59.4516x53.5304,-64.7214x47.0228,-69.282x40,-73.0836x32.5389,-76.0845x24.7214,-78.2518x16.6329,-79.5618x8.36228,-80x0,-79.5618x-8.36228,-78.2518x-16.6329,-76.0845x-24.7214,-73.0836x-32.5389,-69.282x-40,-64.7214x-47.0228,-59.4516x-53.5304,-53.5304x-59.4516,-47.0228x-64.7214,-40x-69.282,-32.5389x-73.0836,-24.7214x-76.0845,-16.6329x-78.2518,-8.36228x-79.5618,0x-80,8.36228x-79.5618,16.6329x-78.2518,24.7214x-76.0845,32.5389x-73.0836,40x-69.282,47.0228x-64.7214,53.5304x-59.4516,59.4516x-53.5304,64.7214x-47.0228,69.282x-40,73.0836x-32.5389,76.0845x-24.7214,78.2518x-16.6329,79.5618x-8.36228,80x0
bed_temperature = 55
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1.1
bridge_speed = 40
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M140 S0; Turn off bed\nM104 S0 ; turn off temperature\nG28 \nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 0
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.5
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_diameter = 1.721
fill_angle = 45
fill_density = 0%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 70
first_layer_extrusion_width = 125%
first_layer_height = 0.2
first_layer_speed = 30
first_layer_temperature = 220
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 3
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 120
interface_shells = 0
layer_gcode = 
layer_height = 0.1
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 45
min_print_speed = 30
min_skirt_length = 15
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 80
perimeters = 1
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 2.5
retract_length_toolchange = 10
retract_lift = 2
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 80
seam_position = aligned
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 30
small_perimeter_speed = 80
solid_infill_below_area = 25
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 1
standby_temperature_delta = -5
start_gcode = M666 Z-.45 Y.6 X-.2\nG28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 220
thin_walls = 0
threads = 3
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 0
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 1
xy_size_compensation = 0
z_offset = 0

teppix, 2015-07-22:
I think that the width of the perimeters are larger in some cases when detect thin wall is enabled. This is meant to fill the small gaps between the perimeters. To compensate for this, the path has to be moved inwards a small amount. Check each layer in the preview to see if the extrusion width varies from layer to layer.

Praesmeodymium, 2015-07-22:
Spiral vase mode.... i'm not sure how this comment applies
it should be a single perimeter size afaik. its a solid object slic3r is spiralizing so the perimeter width is all slic3rs choice. And not sure how to check layers in spiral mode since every z increment is a layer.

luisonoff, 2015-07-26:
So it looks nicer when you disable "detect thin walls"?
When using Spiral mode I see no reason to enable thin wall detection, as there is only 1 perimeter right?
But you have a point there, having it enabled should not affect the print.
For developers, maybe if you select spiral mode this option (detect thin walls) should be automatically disabled?

Praesmeodymium, 2015-07-26:
Yeah I was just looking at this again. 
A single wall of this vase with no top and detect thinwalls is fine.

The issue is specifically spralized + thinwalls and I agree I see no point that it should be enabled with spiral. but I dont know every case either. I only noticed this because I left thinwalls detected when copying a config.

teppix, 2015-07-29:
@Praesmeodymium 
I didn't notice that you used the spiral vase option. I agree that this sounds like weird behaviour.