slic3r/Slic3r #3012: Feature Request: Built in calibration prints

pyrophreek, 2015-07-22:
I was thinking about ways to make calibration easier, and I am wondering if it would be possible to integrate a set of calibration prints, sort of like what you get on inkjet printers for head alignment.

Some ideas for possible test prints:

Retraction: 
User inputs a range of potential retraction values (ie. min 0, max 10)
Slic3r generates several towers across the build area and varies the retraction value either as Z increase or  between different towers.  After print completes, user inputs the height at which strings/blobs disappear for the Z increase or which towers in the array print best for the second option.

Bridging:
User inputs a range of speeds and/or bridge flow ratios 
Slic3r generates two walls and then bridges across them while varying the parameters, then tries to print a number on each one. After the print, user inputs which bridge worked the best.

Filament diameter/extrusion factor:
Not sure of the best way to do this, so open to suggestions.  Perhaps matching solid pins/holes, or solid/single wall cubes.

Maximum print speed/minimum layer time (calibration for autospeed perhaps?):
square tower with speed increasing with Z.
Alternatively, maximum extrusion rate could be calibrated for by printing a single wall cube with increasing speed until the motor starts skipping etc.  The height at which it occurs is then input back into slic3r to calculate the max rate.

Support material contact Z-distance:
Print a large raft, then print an array of towers at various contact heights.  The user then inputs which ones were able to be removed, and any that had poor first layer quality.  

Any other ideas, or suggestions are welcome.

hoh61, 2015-10-05:
Hello,

I'm new in the 3D print community and I'm facing the problem to get reliable basic parameter sets for the print materials. Since many of these parameters are strongly dependent on the material itself, I thought of some basic print tests to evaluate the proper settings.

My first attempts are done outside any silcer software (spreadsheet) but using the math given in the slic3r manual. 

Actually I sat up the following tests:

Filament diameter/extrusion factor:
Since the filament diameter is fixed, the extrusion factor is widely variable. Print result is additionally affected by the extrusion temperature, therefore I sat up a print array of small squares with variation of the extrusion factor. This will be extended with an additional variation of the extrusion temperature too. So for one layer you will get the optimum setting.

Speed:
In theory the viscousity of the print material will limit the amount of material provided during printing. 
I've generated a test structure for one layer where I print lines on a rectangle with maximum print speed but variable acceleration.  With increasing acceleration the maximum speed in the middle of the lines increases. At a critical speed the extruded material is so few that the lines no longer stuck together. The onset of this effect is clearly visible, the location can be measured, the maximum speed can be calculated from this geometry.
(ptoto to be amended)

I have prepared this for the first layer yet, but it can be transferred for subsequent layer too (not yet done.

I have also an idea for the retraction setting test, whcih I will set up wit next days.

pyrophreek, 2015-10-05:
I like your idea for the max-speed test.  One really nice thing about it, is you could use machine vision to evaluate it.  Process could be as follows:
1)print lines at slightly varying height and/or spacing using slow speed and no acceleration until a gap forms between them (this could actually be used as a first layer height precision calibration as well) to make sure that minor changes will be detected. (use camera to detect when this occurs)
2) print your test algorithm and when a certain amount of gap appears, max speed would then be quickly found from a lookup table.

While I think this might be beyond the scope of Slic3r integration at the moment, it could be something to be integrated into future systems.

a4jp-com, 2015-10-06:
That's a nice idea.

lordofhyphens, 2016-05-17:
I don't see this as something that should be part of Slic3r. You are welcome to set up a set of models, etc on Github as a separate project though.

Going to close this. If @alexrj wants to take this on, he can repoen it :)

closed 2016-05-17 by lordofhyphens