slic3r/Slic3r #3013: First layer extrusion multiplier

foulowl, 2015-07-22:
Hello,

Generally the recommendation is to turn up the first layer extrusion width to get the first layer to stick better. However, turning up extrusion width *also changes the path spacing* as described here:
http://manual.slic3r.org/advanced/flow-math
This is both expected and desired, but in my case I need a way to change the total amount of plastic that gets squirted out on the first layer.

I have tried turning up the "extrusion multiplier" and it clearly allows the first layer to stick better. However, this comes with the usual consequences of having too much extrusion for the rest of my print. Ie, little blobs on thin walls, etc.

My solution: Allow an "extrusion multiplier" setting for just the first layer. The current "extrusion multiplier" setting would apply to all layers besides the first layer. Thus, in my case I would set the "first layer extrusion multiplier" to 1.05, and "other layers extrusion multiplier" to 1.0 to get my desired result.

(As a item of note: there is in fact an upper limit on extrusion width, contrary to what is stated here:
http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#Layer_height.2C_Extrusion_width
If you turn up extrusion width too much, your paths aren't connected at the sides. I have found experimentally that I max out at around 0.6 mm extrusion width, which is 150% of my nozzle width. This depends on first layer height as well, but since you will generally want a fatter first layer height, this also limits your extrusion width.)

Thank you!

Mikk36, 2015-07-23:
The other option is to just move your nozzle closer to the bed when being at Z0.

foulowl, 2015-07-23:
Moving the nozzle closer to the bed when at Z0 helps the first layer to stick, but it doesn't accomplish what I am asking here: For the nozzle to extrude more plastic on the first layer only.

Thanks!

lordofhyphens, 2015-07-23:
For most cases you can apply a modifier with a script to increase printing
moves.

I can think of a 10 line script that would handle it assuming that the
first layer is the first set of printing moves.

Gregregorovich, 2018-05-25:
Hi, found this via google (I was looking for increasing first layer extrusion multiplier).

This issue appears to be closed as there's a "First layer" sub-setting of "Extrusion width". [I'm still new to Slic3r, but I think I know what I'm doing : ) ]

-Greg

lordofhyphens, 2018-05-25:
@Gregregorovich the original author wanted to deliberately over-extrude the first layer (I find that this makes the hotend more susceptible to jams myself).

The setting you found tells Slic3r to put out wider traces on the first layer (which often does help with adhesion).

wavexx, 2019-06-14:
I was looking for a way to do this without adding some conditional gcode, but the only way I've found to achieve something similar is by tuning the first layer height, then first layer extrusion width and then set a positive Z offset in order to actually squeeze (errr, overextrude) the layer. Yep, it's actually easier to adjust the extrusion multiplier through some conditional gcode.

So, why would I want this?

I'm trying to produce a textured first layer using a non-conventional technique in which more material is needed as it's being pushed through the ridges of the surface.

You can buy a textured sheet from several vendors now, and I'm also using the stock metal textured sheet from Prusa. These sheets are uniform enough that lowering Z is possible.

However I'm trying to add even *more* texture. I had good results by using 80 grit sandpaper coated, carefully glued on top of a glass sheet and then subsequently coated with high-temp siliconic varnish (talk about overkill). The resulting surface is not as perfect as the textured plates. You cannot lower the nozzle to compensate for the missing material: you really need *more*.

lordofhyphens, 2019-06-14:
If you have volumetric E, you can use a post process script to do the tweak
on layer 0.

At layer conditional gcode is the other route with M221.

wavexx, 2019-06-14:
On Fri, Jun 14 2019, Joseph Lenox wrote:
> If you have volumetric E, you can use a post process script to do the tweak
> on layer 0.
>
> At layer conditional gcode is the other route with M221.

Yes, I'm currently using M221.