slic3r/Slic3r #3014: Feature Request: Coast

wfcook, 2015-07-23:
I've been playing a little with Simplify3D and find it's Coast feature to be quite effective at fighting ooze and stringiness. The feature turns off the extruder for a specified distance before the end of a loop to allow the pressure inside the nozzle to finish it off...I am using 0.6mm and am surprised at how much of a difference it makes. It seems superior to Wipe in that it doesn't smear extra extrusion over the part.

Presumably, one would need to add that distance back in during un-retract but I haven't found that to be necessary, not sure why.

I hope you don't mind my barrage of posts here recently. I really like slic3r and use it daily.

lordofhyphens, 2015-07-23:
@wfcook, I'm glad you lucked out that your flowrate/pressure worked out.
The closest analog to this feature is "Pressure Advance", but it's largely
been superseded by autospeed.

wfcook, 2015-07-23:
I'm using Autospeed in Slic3r but not having as much luck. Not sure why.

luisonoff, 2015-07-26:
That coast feature sounds interesting. I usually print nylon and have lots of trouble with stringing because of the high nozzle pressure it needs. So +1 to this request. :)

wfcook, 2015-07-27:
There really is nothing in Slic3r that I can find that produces this result of letting the pressure off close to the end of a loop. Advance doesn't quite to the same thing, and Coast is superior to Wipe for preventing stringing IMO.

Paul--W, 2015-07-31:
I think Glide will be easy to do, even by post-processing.
What you need to do is first calculate the point to stop feeding and start gliding.  

Please double check my formula:

To calculate a point that is Distance from an ending point along a line:
* For a an example line from (0,0) to (0,10)
x1 = 0
y1 = 0
x2 = 0
y2 = 10

* And we want the coordinates of a point 1 unit short of the end point of that line
distance = -1

* First calculate the vector from x1y1 to x2y2:
vx = x2 - x1
vy = y2 - y1

* Then calculate the line length:
linelen = sqrt((vx*vx) + (vy*vy))

* Normalize the vector to unit length:
vx = vx / linelen
vy = vy / linelen

* Calculate new coordinates, "Distance" from ending point:
px = x2 + (vx * distance)
py = y2 + (vy * distance)

* This yields the point (0,9) where the extrusion would stop, and you would just need to add a move at the original speed from there to the ending point.

Now... by examining the gcode created with autospeed,  it is only making sure that there is not over-pressurization (too many mm3/sec) by slowing down if needed.  There is no extra pressure added at the beginning of a new feature, after an un-retract.

So I propose this for pressurization: 
- Say your printer takes 1mm to get up to pressure.
- For the first segment in the feature (perimiter, infill, support etc) calculate a point along the line 1 mm from the beginning of that feature.
- Print that (1mm) line at a slower - calculated feedrate, to allow pressurization
- Then print the rest of the segment at the original feedrate.

The calculation of the pressurization distance and feedrate needs work, it depends on acceleration, speed, flow rate etc., so might need a formula.  But if you had two constants like Pressurization_distance and Pressurization_speed then you could get close.

Add Glide_Distance and I think you'd have it all.  

What do you think?

wfcook, 2015-08-03:
Sorry for the late reply, I have been at a campground with no internet and barely any phone service.

I had been looking into implementing this also, but while I consider myself a pretty fair C++ programmer, Perl is something I've never worked in, so the learning curve for that has me stalled a bit.

The math looks right, and the concept is sound, but there are a couple of other things we should consider. First, you have to take into account the fact that the coast distance may be longer than the final segment of the loop. Also, what do you do for very short 'loops' that may be shorter than the coast distance.

I'll trey to run a few tests with S3D to see how they handle it.

Bill Cook _/)_

Paul--W, 2015-08-03:
Bill,
Thanks for your reply, and for your hard work.

I'm very interested in this as I have a large Delta (pictured below) that has a long bowden tube (850mm) - so pressurization and coast are features I need in order to print certain objects.  
So I've been working on a gcode post-processor in order to test.  

My post-processor identifies each action in the gcode, and performs various calculations on each segment to determine all aspects, and creates a table that I can easily examine, and perform additional calculations.  For Coast:  
- It determines the total loop length, and will only apply Coast before a retract if the loop length is greater than a minimum.  I am planning on the ability to coast for more than one segment if the segments are shorter than the coast distance. 
- I am considering how the pressure might need to be increased at the beginning of a loop, or if the total extrude is just done in a shorter distance - so total extrude amount does not change.  If the extrude portion is done in a shorter distance then the mm3/sec will change... Still thinking about that - will have to test.

<b>wfcook - it would be very helpful to have an STL and gcode example from a program that does Coast!  Can you upload that?</b>

Other things I'm working on testing:
- Pressurize.
- Changing retracts within support structure to what I'm calling "Moving Retracts" - instead of retracting and then moving, retract while moving in the right direction (with an optional hop up at an angle), unretract while moving (hop down at an angle).  This should speed up support printing without loosing pressure, and I don't care about a little stringing within support. For a full sized helmet print I tried just removing the 25,000 retracts in support - but then there was a loss of pressure.
- Re-pressurize after bridging, since bridging dumps pressure.
  
What it looks like so far (it also plots the gcode, colored by mm3/sec)
[image]

And this is my printer, Raptosaur:
[image]

wfcook, 2015-08-03:
Wow, you are doing some awesome work on this. I will definitely put something together that you can download to get a look at coast. I'm thinking a couple of hollow cubes like this (http://www.thingiverse.com/thing:45225) so the head has to move across a gap between them, and a few samples of generated g-code using different coast settings and speed settings.

What language are you doing this in? Would it be possible, at some point, for me to get a look at it (I understand if not). I am hoping we can eventually gin something up Slic3r itself.

I too have a large delta and this is why I am interested (I know the print bed could be larger, but it suits my needs for now):

[image]

Paul--W, 2015-08-03:
Bill,
Nice printer!
I'm actually writing this in a database language called Microsoft Visual Foxpro 9 (unfortunately no longer available). This allows me to do rapid prototyping and work with the data very easily indeed.  I'll be happy to share the code when it's working, it's easy to read.  The main thing will be to see what works, then we can transfer the concepts.

wfcook, 2015-08-03:
Yup, I knew Visual Foxpro once upon a time, but it's been a long while.

I just had a look at a G-code file I have here, and the concept is unbelievably simple. Basically, it simply stops extruding for the length that you tell it to at the end of a loop (g-code for loop at end of this post). I suppose this could combine that with wipe-while-retracting and possibly extra-length-on-restart, but I was leaning toward doing something more sophisticated to maintain pressure and you are way ahead of me on thinking this through. I like your approach.

This g-code is really simple. Settings for this file were 0.6mm nozzle, 0.64mm extrusion width, 0.4mm layer height, 40 mm/s speed, 1.75mm filament. S3D apparently assumes a rectangular extrusion shape, so it comes out to 0.64*0.4mm = 0.256 mm^2 = mm^3 of filament per mm of travel while extruding. The reduction in stringing is simply astounding for such a small change.

; solid layer
G1 X14.341 Y38.778 F9000
G1 Z0.800 F9000
G1 E0.0000 F3600
G92 E0
G1 X13.680 Y38.117 E0.0995 F2400
G1 X15.356 Y40.699 E0.4271
G1 X16.011 Y41.354 E0.5257
G1 X16.596 Y42.256 E0.6402
G1 X16.727 Y42.474 E0.6672
G1 X16.829 Y42.672 E0.6910
G1 X16.919 Y42.881 E0.7152
G1 X16.992 Y43.089 E0.7386
G1 X17.046 Y43.294 E0.7612
G1 X16.760 Y43.008 E0.8043
G1 X16.811 Y43.138 E0.8192
G1 X16.915 Y43.558 E0.8653
G1 X16.960 Y43.993 E0.9118
G1 X16.956 Y44.109 E0.9241
G1 X17.139 Y44.292 E0.9518
G1 X17.105 Y44.633 E0.9882
G1 X17.047 Y44.917 E1.0191
G1 X17.002 Y45.060 E1.0350
G1 X16.855 Y44.913 E1.0571
G1 X16.673 Y45.414 E1.1138
G1 X16.599 Y45.562 E1.1315
G1 X16.741 Y45.704 E1.1528
G1 X16.529 Y46.121 E1.2025
G1 X16.455 Y46.323 E1.2255
G1 X16.310 Y46.178 E1.2473
G1 X16.221 Y46.420 E1.2747
G1 X16.142 Y46.712 E1.3069
G1 X16.111 Y46.884 E1.3255
G1 X16.273 Y47.047 E1.3500
G1 X16.236 Y47.460 E1.3941
G1 X16.250 Y47.902 E1.4412
G1 X16.255 Y47.933 E1.4445
G1 X16.054 Y47.733 E1.4747
G1 X16.061 Y47.894 E1.4919
G1 X16.116 Y48.296 E1.5350
G1 X16.208 Y48.674 E1.5764
G1 X16.278 Y48.862 E1.5978
G1 X16.617 Y49.200 E1.6487
G1 X16.628 Y49.226 E1.6518
G1 X16.839 Y49.593 E1.6968
G1 X17.093 Y49.936 E1.7422
G1 X17.396 Y50.258 E1.7893
G1 X17.741 Y50.545 E1.8370
G1 X18.120 Y50.790 E1.8851
G1 X18.288 Y50.872 E1.9050
G1 X18.644 Y51.227 E1.9585
G1 X18.891 Y51.310 E1.9863
G1 X19.277 Y51.396 E2.0284
G1 X19.665 Y51.442 E2.0699
G1 X19.766 Y51.444 E2.0807
G1 X19.561 Y51.239 E2.1115
G1 X19.831 Y51.263 E2.1404
G1 X20.271 Y51.248 E2.1873
G1 X20.448 Y51.221 E2.2063
G1 X20.612 Y51.385 E2.2310
G1 X20.763 Y51.361 E2.2473
G1 X20.875 Y51.335 E2.2595
G1 X21.187 Y51.243 E2.2941
G1 X21.321 Y51.189 E2.3095
G1 X21.177 Y51.046 E2.3311
G1 X21.561 Y50.886 E2.3754
G1 X21.791 Y50.754 E2.4036
G1 X21.932 Y50.895 E2.4247
G1 X22.240 Y50.691 E2.4640
G1 X22.456 Y50.514 E2.4938
G1 X22.322 Y50.380 E2.5140
G1 X22.449 Y50.272 E2.5317
G1 X22.747 Y49.961 E2.5775
G1 X22.773 Y49.926 E2.5822
G1 X22.906 Y50.058 E2.6021
G1 X23.066 Y49.860 E2.6292
G1 X23.275 Y49.544 E2.6696
G1 X23.282 Y49.530 E2.6712
G1 X23.140 Y49.388 E2.6926
G1 X23.224 Y49.244 E2.7103
G1 X23.355 Y48.960 E2.7436
G1 X23.422 Y48.764 E2.7656
G1 X23.571 Y48.914 E2.7881
G1 X23.602 Y48.837 E2.7969
G1 X23.712 Y48.457 E2.8391
G1 X23.759 Y48.197 E2.8672
G1 X23.595 Y48.032 E2.8919
G1 X23.627 Y47.653 E2.9324
G1 X23.611 Y47.220 E2.9785
G1 X23.598 Y47.131 E2.9882
G1 X23.806 Y47.338 E3.0194
G1 X23.804 Y47.276 E3.0260
G1 X23.755 Y46.889 E3.0676
G1 X23.668 Y46.509 E3.1091
G1 X23.575 Y46.231 E3.1402
G1 X23.560 Y46.187 F2400
G1 X23.169 Y45.796 F2400
G92 E0
G1 E-4.5000 F3600
G1 Z0.900 F9000
G1 X-8.572 Y45.734 F9000
G1 Z0.800 F9000
G1 E0.0000 F3600
G92 E0

lordofhyphens, 2015-08-03:
@wfcook, most of the guts of Slic3r these days is C/C++ (via XS); I think
@alexrj just moved more of the gcode writer to C++.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Paul--W, 2015-08-03:
Bill,
Yes that is really simple :)
I ran that gcode through my analyzer and the mm3/sec for all the extrudes is virtually identical (10.24 mm3/sec), so they are not doing anything fancy at the beginning of the loop/layer, which had a total length just over 29mm.
What I think is happening is that under normal circumstances the retract in effect stores the pressurization, you get it back on unretract as long as it was retracted far enough that it doesn't ooze.  

I will implement coast as I outlined for now, without worrying about the more complicated stuff.  Simple is better, and I have a simple test in mind.

It may need a little added extrude amount at the beginning - the amount that was not extruded in that last .65mm perhaps.

Paul--W, 2015-08-04:
Bill,
I have "Coast" programmed in the post-processor, everything went as expected (programming wise).  
I need to plot it out and make sure my math is working.  Note that I truncate the extrude decimals to 3 places as that is below stepper resolution, and do not specify the feed-rate unless it changes.  I'm working with 112MB gcode files (verbose mode on), so I shrink the output file size when I can.

Original:
G1 X-37.685 Y-98.850 E0.977  ; normal extrude
G1 X-38.071 Y-97.564 E1.001  ; this will be broken up into a partial extrude and a coast
G1 X-38.584 Y-97.519 E1.011  ; this will be a Coast
(retract)

New with 0.7mm Coast:
G1 X-37.685 Y-98.850 E0.977   ; normal extrude
G1 X-38.018 Y-97.741 E0.138   ; Part Extrude before Coast
G1 X-38.071 Y-97.564               ; COAST
G1 X-38.584 Y-97.519               ; COAST
(retract)

wfcook, 2015-08-04:
lordofhyphens, I am aware that most of Slic3r is C++, but I was under the impression that features such as this one were supposed to be implemented in PERL. Or is that incorrect?

wfcook, 2015-08-04:
Paul-W, let me know how it goes. As an aside, what hotend are you using with your long bowden?

Paul--W, 2015-08-04:
Sure, I'll be working on it this evening.

It's a hotend I designed and built myself from tellurium copper, some nichrome from a fish tank heater, stainless tube, and Autocrete- Hi-Temp Ceramic Adhesive, with some machining help from my brother.  
It has an .571mm orifice.

[image]

[image]

wfcook, 2015-08-05:
That's really cool. No end of problems here with the current hotend jamming...going to have to make a switch I think

lordofhyphens, 2015-08-05:
@wfcook, what hotend are you using? I know that the cheap knockoffs are
extremely questionable w/r/t quality.

You might find some other recommendations in #reprap on freenode irc

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Paul--W, 2015-08-06:
It will be a day or so more before I can publish the Coast methods.  Just put Win 10 on laptop, testing.

wfcook, 2015-08-06:
No rush, thanks

Paul--W, 2015-08-16:
This is a translation to pseudo code of  what I am doing to post-process gcode to include Coast segments, kind of.  The technique in Slic3r would be necessarily a bit different, I have no idea what kind of data structures are used.

I actually read the entire gcode into a table with 33 extra fields - used to calculate many things like: 
segment length, total loop length, segment type, mm/second, segment time, segment extrude mm, segment extrude mm3, segment mm3/second, etc. for debugging.  I've got the post-processor done, and I will try to do some test prints this week with the test object below that has various sized lines from 10mm to 60mm. 

Notes:
* The Gcode I have in the post a few posts up was incorrect.  I had reversed some coordinates in a calculation.
* I use the Verbose G-Code option under "Print Settings", "Output Options" in SLic3r to make it easier to identify different kinds of segments.
* [Treat statements like these as pseudo code]  

<pre>
********************************************************************************
* Your Coast settings (arbitrary values here)
* Loops must be at least this long in mm (ending in a retract) before coast is used:
My_Minimum_Coast_Loop_Length = 15      
My_Coast_Distance            = .6

[Iterate through each line in the Gcode from top to bottom (Scan-Loop-1)]
    [if the segment is a move or extrude segment (has X and Y) then save the lastX and lastY corrdinates] 
 
    [if not a printing segment (no E parameter) then skip to the next segment and return to top of Scan-Loop-1]
    
    * -- Now positioned at the first printing segment in a loop -- *
    this_loop_length = 0
    found_a_retract  = 0
    found_a_move     = 0
    
    * -- FIND A LOOP OF THE RIGHT MINIMUM LENGTH, ENDING IN A RETRACT -- *
    [Iterate through each line in this loop]
        [if the statement is a retract]
            found_a_retract = 1
            [store the retract_position in the gcode]
            [exit from loop]
        [endif]
        [if the statement is a move (non-printing)]
            found_a_move = 1
            [exit from loop]
        [endif]
        
        [Extract the X and Y coordinates to Xpos and Ypos]
        
        * -- CALCULATE THIS SEGMENT LENGTH -- *
        this_seg_length  = abs(sqrt((Xpos-lastX)^2 + (Ypos-lastY)^2))
        
        * -- ADD TO THE TOTAL LOOP LENGTH -- *
        this_loop_length = this_loop_length + this_seg_length

        [store Last_Printing_Segment_Position]

    [End]

    [if found_a_move = 1 or found_a_retract = 0 or this_loop_length < My_Minimum_Coast_Loop_Length]
         * -- NOT THE KIND OF LOOP WE WANT -- *
         [return to top of Scan-Loop-1]
    [endif]

    
    * -- FOUND A LOOP OF THE RIGHT MINIMUM LENGTH, ENDING IN A RETRACT -- *
    
    [go back to Last_Printing_Segment_Position]

    Remaining_Coast_Distance_for_Loop = My_Coast_Distance

    [From Last_Printing_Segment_Position towards the top, while Remaining_Coast_Distance_for_Loop > 0    (Scan-Loop-2)]
        If segment_length <= Remaining_Coast_Distance_for_Loop
            * -- This is a short segment at the end of the loop, change it to a move (coasting) -- *
            Remaining_Coast_Distance_for_Loop = Remaining_Coast_Distance_for_Loop - segment_length
            [Change the segment to a move by removing the extrude parameter]

         else
         
            * -- THE SEGMENT_LENGTH IS > REMAINING_COAST_DISTANCE_FOR_LOOP -- *
            * -- CALCULATE THE POSITION OF THE NEW END OF THE (SHORTENED) EXTRUDE SEGMENT -- *
        
            * -- DistanceFromEnd is used as a negative number -- *
            DistanceFromEnd = Remaining_Coast_Distance_for_Loop * -1

     
            * -- To calculate a point that is a distance from an ending point along a line:
            * -- First calculate the vector from (lastX,LastY) to (Xpos, Ypos) -- *
            vectorX = lastX - Xpos
            vectorY = lastY - Ypos

            * -- Then calculate the original line length -- *
            OriginalLineLen = sqrt(vectorX^2 + vectorY^2)

            * -- Calculate new coordinates, "DistanceFromEnd" from ending point using (vectors normalized to unit length) -- *
            newX = lastX + ((vectorX / OriginalLineLen) * DistanceFromEnd)
            newY = lastY + ((vectorY / OriginalLineLen) * DistanceFromEnd)

            * -- CALCULATE NEW SEGMENT LENGTH -- *
            newSegLen = abs(sqrt((newX - Xpos)^2 + (newY - Ypos)^2))

            * -- CALCULATE NEW SHORTER SEGMENT % OF TOTAL LINE LENGTH -- *
            newSegPct = newSegLen / OriginalLineLen

            [extract the extruder position of the previous segment to variable "lastExtrudePos"]

            [extract the extruder position of the this segment to variable "ExtrudePos"]
            
            * -- Calculate the extrude length of this segment -- *
            eLenMM = ExtrudePos - lastExtrudePos

            * -- Calculate new shorter Extrude Length based on the new sorter segment length -- *
            newELen = eLenMM * newSegPct
           
            * -- Calculate new extruder pos, shorter by the amount not being extruded -- *
            newEpos = ExtrudePos - (eLenMM - mnewELen)
        
            * -- CHANGE SEGMENT TO A SHORTER EXTRUDE (your concatination code will vary) -- *
            [FixedSegment = 'G1 X'+ newx + ' Y' + newY + ' E' + newEpos + ' ;Part Extrude distance ' + mnewseglen + 'mm']
            [overwrite the current segment with the FixedSegment]
  
  
            * -- CREATE A NEW COAST SEGMENT - A MOVE TO (Xpos, Ypos) WHICH WAS THE ORIGINAL END POINT -- *
            [NewSegment = 'G1 X'+ Xpos + ' Y' + Ypos ]
            [insert the NewSegment after the original segment]
            
            [go back to the retract_position]
            
            [exit from Scan-Loop-2]
        endif
        [skip to previous segment]
    [End of Scan-Loop-2]
    [Skip to next segment]
[End of Scan-Loop-1]
</pre>

[image]

Paul--W, 2015-08-18:
This is what the coast segments look like (blue here) in my debugger.  Work is crazy busy but I hope to test this week.
[image]

wfcook, 2015-08-18:
This looks really good. I'll get back to seeing how I would implement this in the actual Slic3r code. Let me know how the test works out.

Now if they can just fix gap fill...

Paul--W, 2015-08-22:
Bill,
I've done my testing.  I got some interesting results that revealed some deeper issues with long bowden tubes.

My printer has a very long (850mm) bowden tube, with 3mm filament, so it will show these issues more extremely.   My retract distance is 7.8mm at 77mm/sec, and retract could be even a bit longer to eliminate more stringing - which is minimal.

First I tested printing the straight lines without any Coast, and adjusted the "extra-extrude after retract" in my Smoothie firmware from 0 until the beginning of the lines were almost OK.  I used .4mm extra extrude.  This under extrude at the start is due to a pressurization issue from slight oozing (I could retract farther) and the time that it takes for full re-pressurization to occur after un-retract.  This is a problem that gets worse with multiple retracts.  Adding extra extrude after a retract is a bad band-aid, but that's another issue.
  
The second biggest issue I see in the original is the blob at the end of each line where movement stops and a retract happens.

Then I tested printing with increasing Coast distances, up to 15mm of Coast.  At longer Coast distances (where any difference is noticed) it only minimally decreases the blob at the end.  A longer Coast also depressurizes the extruder, so the beginning of the next line is badly under-extruded and the entire extrude width is less - this shows up especially in the short lines.  Note that the amount of filament not fed while coasting should be put back when un-retracting, so under-extrusion does not occur.

I think Coast would work best with a non-bowden extruder, because:
- With a bowden the long spring of filament is keeping most of the pressure up while coasting,
- Coast doesn't help with the longer pause at the end of the extrude (necessitated by the long spring of filament that must be retracted). 

The path I am going to take next is to:
- Make most of the retract happen earlier, x mm from the end of the loop while coasting to the end.  
- If the distance is not enough then the remainder of the retract would have to happen when the motion stops (not ideal, do it earlier!). 
- This should eliminate the blob at the end, most/all of the pause, and there should be better pressure at the start of the next loop than the Coast I tried (especially if I add to the un-retract distance).  
- This should make printing faster too.  

I'll post on the Reprap Slic3r forum when I've got more results.  I'll post back here when I've got a feature request I've proven.   Now that my post-processor is working, there is a lot more I'lll be testing: Pressurize, Proportional Retracts, Moving Retracts in Support, simply slowing the un-retract speed...

This photo has the original print behind, showing slight under extrusion at the beginning which I see when I have multiple retracts, and the blob at the end where the retract is.
[image]
In front is one of the 60mm long lines with 15mm of Coast - showing extreme depressurization at the beginning of the line, and still having a blob at the end.

Thanks, and I hope to work with you again.
-Paul

luke321, 2015-08-22:
Hello Paul,

I am working on my own GCode Post Processor right now, and I want to change the way the printer accelerates and deccelerats at the beginning and end of each line to a more continues way trough small moves with different speed which match a trapezius or sin^2 acceleration.
I think combined with a sort of advanced algorithm this would allow the extruder to pressurize and depressurize while keeping retracts fairly low and print speeds high (except for the beginning and end).
Since your Post Processing script is already working, maybe you could try tingeling with the speed to get better results.

Paul--W, 2015-08-22:
Luke,
I like the idea, and will be looking into that after I have tested what I'm working no now:
<b>Retract and Coast</b> with these features:
- Begin retracting x mm from the end of the loop, while coasting to the end.  Finish retract there if not complete.  Amount not extruded will be added to the un-retract.
- Proportional retraction - more retract for long distance moves to fight oozing.  For instance if moving 10 mm after retract it would retract 5mm, but for a 150mm move it would retract 8.5mm.  This will be done after the first part is complete and tested.
- Hops added to moves (I know Slic3r has this, but I will be doing other special things while hopping)

After thinking about this, I believe the blob that happens at the end while retracting is what is causing the pressure loss that happens with lots of retracts.  I'm hoping this will eliminate most of that issue.  
Then I will turn my attention to pressurizing the start of the extrude - I don't think Advance works properly with long bowden tubes as the in-and-out pulse happens too quickly. A proper algorithm like you are talking about will probably be needed to make the extrusion width come out even.  The first thing to look at is the unretract speed - have a fast start, and slow end perhaps.  We'll see.  I don't want to co-opt this thread any further.

In progress:
[image]

luke321, 2015-08-22:
Very nice, for the proportional retraction, maybe you could retract continously while moving to the next position, I think minimizing rapid extrusion / retraction is they key to faster prints with high quality. 

Looking forward to your plugin.

Paul--W, 2015-08-23:
Luke,
Yes, another feature planned is retract while moving - within support.  Unfortunately I'm not writing a plugin,  this is an separate program that is a gcode debugger, visualizer, and post-processor that I'm creating to understand what's happening and to test methods that can improve Slic3r.  I will publish the methods here however when I've proved them.

wfcook, 2015-08-24:
I was going to suggest a couple of things, most of which have been covered here. This has been a worthwhile test so far I think.

luke321, 2015-08-24:
Hey Paul,
I am doing the same thing actually, reading in the GCode Creating Segments, Clusters and Layers than modifying things, up till now I implemented 2 things:
Slow down of very short travel moves < 2mm which removes a lot of Vibrations. It seems the firmware has trouble applying acceleration correctly to very short moves.
Adaptive Z Lifts: Only do Z Lift for moves > 30mm (or any other value). Which removes a lot of unnecessary z lifts and trough that save time.
Maybe we should create a new issue with our findings, ideas?

Paul--W, 2015-08-24:
Bill,
I want to thank you again for bringing up this subject.   I'm learning a lot.
I have finished testing Retract while Coasting. Here are my final settings:

[image]

Coasting while Retracting works very well to virtually eliminate the blob at the end.  If too much coast distance is used while retracting then the segment is truncated, if too little coast then the blob is bigger.  For my printer:  1.3mm from the end was about perfect for both 50mm/sec and 80mm/sec printing speeds.  

I calculated how much retraction could happen in the time the coast segment happens, and made the coast segment retract that amount (converts additional segments as needed).  Any left over retract was done as the original - when movement has stopped at the end of the segment.  The resulting gcode looks like this:

<pre>G1 X-28.421 Y-14.316 E0.142 F3000 ; perimeter
G1 X-29.721 Y-14.316 E-3.238 ; RETRACT E-3.38 COAST 1.3mm
G1 E-7.158 F7800 ; FINISH RETRACT E-3.92 
G1 Z3.400 F16800.000 ; move to next layer (15)
G92 E0 ; reset extrusion distance
G1 X30.499 Y6.004 F16800.000 ; move to first perimeter point
G1 E7.731 F5400 ; Unretract 7.3 + Not ext 0.021 + Add MM 0.41
G92 E0 ; reset extrusion distance</pre>

Now the pressurization problem at the beginning of the segment was slightly less, but not as much less as I expected!  I think that the problem mostly comes from ooze.  

I tuned retraction using small (6mm) through large (10mm) retraction amounts, and increased my retraction speed and acceleration. With a large retraction I get no stringing, but the ooze still happens.  So I adjusted retraction up till I had no stringing at 7.3mm.

Then I tuned  Extra length on restart  ("Add to Unretract for ooze" in my program), by increasing it until the thickness of the beginning of the line was about the same as the ending.   Unretract speed didn't seem to matter.  This extra extrude came out to about the same as I had in firmware before: .41mm.

Then I compared the thickness of the beginning of each line and found that the lines with longer travel distances were slightly starved at the beginning, short travel = slightly over extruded.  So a longer distance = more ooze.  I plan on adding Proportional Ooze Compensation - that's my next step. 

Luke,
I like what you're doing.  I think you should add a new issue for slowing the short fast moves.

luke321, 2015-08-25:
Hey Paul,
when you write ooze, you actually mean material oozing out of the nozzle while doing travel moves?

Paul--W, 2015-08-25:
Yes.  Ooze is the material that comes out after retraction - there is no pressure but gravity causes the remaining melted filament at the bottom of the nozzle to slowly ooze.  Increasing the amount of retraction will not help this (it's caused by the nozzle geometry). 
This ooze deposits on the outside of the next object (if no z hopping is used) and may look like blobs on the outside if the seam position is random, or may look like a thin diagonal that builds back towards the travel direction if the seam position is consistent.
I have finished coding Proportional Ooze Compensation, and a couple other things (like Wipe and Prime for a clean start, Z hops), and will be testing all this shortly and will report back.

luke321, 2015-08-25:
Okay, I just wanted to clarify, I am using a E3D V6 right now and there is almost no oozing, but I should probably also factor this is for my algorithm. Thx for the feedback.

Paul--W, 2015-08-25:
Yes, could be, or could be both... Hard to say.  I know if I have retracted 10mm then there isn't much filament in the melt chamber, and it still oozes. I am printing at a fairly high temp usually in order to print very fast, and that doesn't help.

wfcook, 2015-08-26:
Wow, this is really great stuff, and I want to thank you for doing the testing!

I will try to set up my development environment here so I can get something usable into slic3r and work my way towards a pull request.

I think the oozing is in part caused by gravity. I know that I get oozing of this type post-retraction as well, but I get more of it with the E3D volcano than the ordinary E3D, which would indicate that it is more dependent on volume of melted material left in the nozzle.

Paul--W, 2015-08-26:
Bill,
I spent quite a bit of time testing, and tweaking, primarily with Ooze Compensation.
- I found that it does make a difference being able to vary the amount of Ooze Compensation - adding a bit more for longer moves helps.  It is difficult to test this (and more needs to be done) as Slic3r doesn't follow the same paths on every layer for models with a large number of parts - it's not using the shortest path.
- This test also exposed the missing piece - the acceleration after un-retract.  By default the quick acceleration at the beginning causes the start of the segment to be starved. As a quick and dirty (and temporary) fix I added a "Dwell after Un-Retract" of 35ms.  This does not solve the problem but does help a lot.  When tuned - there is a very slight blob at the start, followed by a slightly thinner section for a couple mm, and then normal flow.  So acceleration will be needed here, or start the un-retract earlier, or both (I will be testing this).  (The Advance method does something different - it does a pulse and instant retract - which I do not think is best.)  We are moving toward full pressure control.
- Here is the results from the final test:
[image]
There is still a slight blob at the end, but if I increase the coast distance then the segment is truncated.

- Here are the settings now, with the new Dwell after Un-retract (band-aid).
[image]

- Here is an example snipet:
<pre>G1 X-16.711 Y-58.01 E6.781 F2700 ; 49.914mm skirt
G1 X-18.011 Y-58.01 E3.026 ; RETRACT E-3.756 COAST 1.3mm
G1 E-0.779 F7800 ; FINISH RETRACT E-3.804
G0 Z0.95 F9000 ; Z Hop
G92 E0 ; reset extrusion distance
G1 X5.882 Y-65.832 F16800.000 ; 25.141mm move to first brim point
G1 Z0.4 F9000 ; UnHop
G1 E7.964 F5400 ; Unretract 7.56 + Not ext 0.039 + Add MM 0.365
G4 P35 ; dwell ms 
G92 E0 ; reset extrusion distance
G1 X5.882 Y-126.496 E1.83370 F2700.000 ; 60.664mm brim
</pre>

If you're interested I'll keep going here.

luke321, 2015-08-26:
Very nice! I have spare time the next 4 days so i will try my version of pressure control and post back here.

Paul--W, 2015-08-26:
Luke,
I look forward to it!

wfcook, 2015-08-26:
I really feel like we're making some advances in the state of the art here. And of course when I say 'we' I mean 'you', but I want to bask in the glory anyway :-)

wfcook, 2015-08-26:
And keep going! By all means keep going!

wfcook, 2015-08-29:
I had a thought this morning...I wonder what the result would be if coast were measure not in distance of head travel but instead in length of extruded filament...i.e. don't extrude the last x mm of filament in each loop.

lordofhyphens, 2015-08-29:
Gets more complicated once you are looking at volume extrusion.

wfcook, 2015-08-29:
Good point

lordofhyphens, 2015-08-29:
Additionally, the maximum amount of coast you get is roughly equivalent to
the mm^3 that's in the melt chamber. At least it seems that way to me, I
haven't thought about it that much.

I wonder if, when looking at the algorithm that's been cooked up it might
be more straightforward to bake it into firmware, since it seems that many
of the tunables are printer-dependent (and the printer fw has more
information about what is actually going on than the slicer or output
gcode).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

wfcook, 2015-08-29:
I don't think firmware is the place for it. For one thing, the correct coast seems in my experience to be speed dependent. It also depends on nozzle size and melt chamber size. I suppose that not everyone has quick-change print heads, but I do. I don't like the idea of having to change firmware settings every time change nozzles, and certainly not every time I change print speed.

wfcook, 2015-08-29:
Actually, this doesn't really get more complicated with volumetric extrusion, as that mode is simply a calculation that converts to a conventional extrusion rate based on speed. The Coast calculation works normally after that.

Paul--W, 2015-08-31:
I spoke with a friend that has S3D and hadn't used Coast on his non-Bowden printer.  He tried it and said it was magic, it really cut down on the stringing.

I think it is very important to have the settings easy to tune and understand, and definitely quick to change.  Having it in firmware would at least require a printer reset (if you have a Smoothie) or worst case - rebuilding and uploading the firmware, and so would make it much harder to tune. 

I'm still working on testing and debugging what I've done so far, and working on what happens when doing the un-retract and start of segment, and better test cases.

lordofhyphens, 2015-08-31:
@Paul--W w/r/t firmware: that's why we have space in eprom for the most basic case. Smoothie just does it even better :)

Paul--W, 2015-08-31:
@lordofhypens   Yes and indeed. 
And things may be even "more different" than we realize once more experimentation is done. 
I just did a test of a complicated pot that had > 50 retracts in a row close together, and it over extruded those, so I have more tuning to do.  I also looked closely at the debug output I found that one of my factors wasn't adding properly, so it's time to simplify the code before the next step.

wfcook, 2015-08-31:
@Paul--W I am glad for the affirmation on the value of S3D's coast feature. The term "magic" is apt.

@lordofhyphens w/r/t firmware: I'm not a particular fan of ooze control as a firmware feature unless the firmware is capable of handling it in a smarter fashion than is likely to be put into g-code. It is just too dependent on features that are likely to change frequently, like the material being used or the nozzle diameter. I don't want to bake that sort of thing into firmware.

wfcook, 2015-08-31:
As I have noted in issue #2960 , there is definitely a problem with Gap Fill and extrusion calculations in general in 1.2.9.x, so @Paul--W be sure that isn't an issue with your testing.

luke321, 2015-08-31:
@Paul--W I think to print lot's of retracts in a row close together we need to reduce speed to get good results, I am working on my code base, mostly some gui stuff and saving settings and hope to get some speed features ready by the end of the week.

Paul--W, 2015-08-31:
@Luke
  Great!

a4jp-com, 2015-09-01:
Are the changes here for Slic3r or a new type of program? I'm happy to test or try anything as long as it doesn't break my printer.

Paul--W, 2015-09-01:
@a4jp-com 
After these experimental techniques are found to work (more testing and development is needed) we are planning on adding them to Slic3r, however nothing is ready to test yet, stay tuned.

a4jp-com, 2015-09-01:
Very nice. I just thought @alexrj was the only developer that could make changes to the software.

wfcook, 2015-09-01:
@a4jp-com Yes final approval requires @alexrj, but we can submit pull requests so that he can integrate features we develop if he so chooses.

a4jp-com, 2015-09-01:
Okay ^^ I hope he comes back soon.

Paul--W, 2015-09-01:
@wfcook 
Regard problem with Gap Fill.  I just re-read that.  Thanks, I will keep that in mind.  
I found the reason my last test with multiple retracts were so over extruded - I still had hardware retracts in the gcode for just short segments I was ignoring - and the firmware was adding extra filament on un-retract.  I need to make sure I do it all one way. I will test again in a few days after some more coding.

[29 more replies]