slic3r/Slic3r #3019: Avoid Crossing Perimeters: Not very effective?

CCS86, 2015-07-26:
Hi, I am running Slic3r 1.2.9 and am not seeing  a very effective result from the avoid crossing perimeters function. I have a print that could pretty easily be done without ever crossing a perimeter, yet a have many, many moves that do so:

[image]

Without that setting enabled, you can see an increase in these moves, but it isn't very dramatic:

[image]

It seems like there is room to optimize that algorithm.

Thanks!

SharkAttackOmNom, 2015-07-26:
Try out various Seam Positions, between Aligned, Random, and Closest. (closest would probably provide the desired result.

CCS86, 2015-07-26:
Thanks for the suggestion. I did try that, but saw very little change.

CCS86, 2015-07-26:
Here it is set to nearest: 

[image]

SharkAttackOmNom, 2015-07-27:
While I understand the desire for improved code, lets try out some other work-arounds. It looks like these are all because of the infill patern and Slic3r probably jumping from optimal Infill paths. Doing Rectilinear paterns through a circular object can be a bit..cumbersome. 

Try out other infills and even numbers of shells. Setting the shells to a ridiculous number (like 1000) might cure the issue entirely.

CCS86, 2015-07-27:
I am happy to test other infill patterns; what do you recommend? 

I want a sparse infill, so fudging the number of shells isn't an option.

Rectilinear way not be ideal on a circular object, but it should still work. Cura can slice with rectilinear infill and has very few moves through the interior:

[image]

gege2b, 2015-07-27:
maybe try with the "concentric" infill pattern, playing with the density to have the sparse infill you look at

CCS86, 2015-07-27:
Concentric infill just isn't going to give me the result that I want. I need to have some strength.

c-born, 2015-07-28:
As I recall the last version where Avoid Crossing Perimeters actually worked properly was 0.9.10b
You could give that a try, although you lose a lot of new features.
I'd be interested in seeing the result

CCS86, 2015-07-29:
It's good to know that it was previously working correctly.

I hope it is acknowledged as a bug for future repair.

krizzap, 2015-07-31:
I agree 1000% this needs to be fixed, its beyond irritating that this feature doesn't hardly work, and fudging other things isn't and shouldn't be the answer. Not to go there but Cura does a good job of this as stated above, why can't Slic3r?

CCS86, 2015-08-17:
Hmm, Slic3r dev is a little sleepy these days.

lordofhyphens, 2015-08-17:
It comes and goes.

RaabenFuchs, 2019-03-09:
Hello everyone, i'm joining this issue, because i print a power supply case at the moment with vertical ventilation holes-> many islands

this causes ridiculous travel moves (every island one time across the whole surface, half arround the whole object, and then a y axis move across the wall) arround the whole object!
(win 64 v1.3.0)

hope you can estimate the moves a bit, really funny:

[image]

i'm total newby, but really prefer  slicer over cura! thx a lot