slic3r/Slic3r #3020: Latest git pull hangs in tests...

wolfmanjm, 2015-07-26:
The latest git pull hangs while doing tests...

t/retraction.t ................. ok     
t/shells.t ..................... 

Linux Ubuntu 14.04 132x also 64x.

atupek, 2015-07-28:
I am also having this problem, hangs on the same test.  Also using Ubuntu 14.04 64x.  Using Perl 5.14.4

mikenew12, 2015-07-28:
Same behavior on OS X 10.10.5. 

Removing the ```shells.t``` test causes it to hang on the ```skirt_brim.t``` test, and removing that test will let it finish the build. So whatever the issue is, it's shared between those 2.

llluis, 2015-08-10:
This was corrected in the last commit: https://github.com/alexrj/Slic3r/commit/5b8ed7367a2a9148c215eafa53c7e13b30cd0c36

closed 2015-08-10 by wolfmanjm