slic3r/Slic3r #3021: volumetric (automatic speed) uses first layer width to set speed for rest of layers

wolfmanjm, 2015-07-27:
I have automatic speed set to 25mm³/sec maximum and maximum speed set to 50mm/sec.
This works well when the first layer width is the same as the other layer widths.
However if I set first layer width to be wider than the rest of the layers it appears the speed on the rest of the layers is being set as if the width were the first layer width instead of the actual width for the layer it is printing.

So first layer width is 1.5mm and layer ht 0.2, then followin glayers the width is 1mm and height 0.5mm. this should have a speed around 50mm/sec. Instead the speed is being set to  around 30mm/sec, as if the width were 1.5mm instead of 1mm

wolfmanjm, 2015-07-27:
Update.. It seems that the first layer height is also affecting it, but no the way I expected.
Setting first layer width back to the same as the other layers had little effect, the other layers were still going half the speed they should have. Setting the first layer height though to the same as the other layers height the speeds now went back to the expected speed (about twice as fast).

I had first layer height set to 0.2 and subsequent layers set to 0.5. setting first layer height to 0.5 fixed the speeds. Unfortunately I need to be able to set the first layer width and height differently to the other layers to get decent adhesion.

I'm not sure there is enough information here to debug the issue, so let me know more you need :)

wolfmanjm, 2015-07-28:
Correction on further testing it appears only the first layer height setting is causing incorrectly slow speeds on subsequent layers, not the first layer width.
I also tested on 1.2.9 stable release and the same thing happens.

Set first layer ht to the same as layer ht, and the correct extrusion speedis generated based on the 25mm³/sec se tin advanced speed setting.
Set first layer height to 0.2 and layer height to 0.5 and all layers are printed at a speed aboot half of what it should be doing.

alexrj, 2016-03-13:
I confirm this behavior. We should probably recalculate it for each layer (region).

alexrj, 2017-03-02:
Okay, autospeed is not recalculated for each layer.

alexrj, 2017-03-02:
Okay, autospeed is now recalculated for each layer.

closed 2017-03-02 by alexrj