slic3r/Slic3r #3022: draw Preview and Layers with selected Z after option change

efa, 2015-07-27:
Open an STL, lets slice it.
Go to a Plater,Preview tab and select an intermediate Z level
Go to a Plater,Layer tab and select an intermediate Z level
Go to Print Settings, change Infill or Support material settings (or some other)
Back to Plater tab,Preview tab, the right Z slider is set to the previous level, but the Preview show Z=max
Back to Plater tab,Layer tab, the right Z slider is set to the previous level, but the Preview show Z=max
Moving the Z slider a little and back, re-draw the selected layer.

Please show the current selected layer after an option change, this is very usefull to compare the effect of Infill/Support setting change
Slic3r 1.2.9 on Ubuntu

Would be nice to have an option to select Z slider unit settings, to mm or layer number.

teppix, 2015-07-30:
Looks like a duplicate of https://github.com/alexrj/Slic3r/issues/2963 (except for the z-slider unit request)

alexrj, 2017-03-20:
This was done a long time ago. Thanks :)

closed 2017-03-20 by alexrj

efa, 2017-03-20:
done in the git sources, but not released?
For the RFE: Z slider unit settings (mm or layer number), do you prefer a new issue?

lordofhyphens, 2017-03-20:
@efa get a build from http://dl.slic3r.org/dev

Prefer a new issue for separate things, check to make sure it isn't already in the issue tracker though.

I don't think having the slider in terms of layer # makes any sense at all once  you realize that not everything is necessarily on the same layer (combined infill, any support, etc).

efa, 2017-03-21:
thanks for pointing me to prebuilt dev versions, I will test.

For the RFE, done in #3783

lordofhyphens, 2017-03-23:
Delete the entire dir and check it out again. Do not use sudo to build.

efa, 2017-03-24:
Open an STL, lets slice it.
Go to a Plater,Preview tab and select first/bottom Z layer.
Go to Print Settings, change Infill, Bottom settings
Back to Plater tab,Preview tab, the right Z slider is set to last/upper layer, and Preview show Z=max
Moving the Z slider back to first/bottom Z layer, show the layer with the new settings.

Please keep the Preview selected Z layer for Infill, Bottom settings too.
I can confirm it is solved for Layers tab with 1.3.0 dev