slic3r/Slic3r #3023: 1.2.9 Autospeed is not respecting Max Volumetric Speed

Paul--W, 2015-07-30:
AutoSpeed is not respecting Max Volumetric Speed

I created a simple object with support.
I have set max speed of 75mm/s and max Volumetric Speed of 18 mm3/s.
When printing some perimeters the actual mm3/sec is over 25.  This often happens with the first perimeters over the support interface.

I calculated this using a Gcode debugger I wrote to help with this, that calculates all parameters, and plots the gcode in color according to cubic mm/sec extruded.  This would be a handy feature in Slic3r...

I have Windows 7 Pro.

This is a 3D view in Slicer:
[image]

This is a layer showing in my debugger indicating in red, the segments that will print at over 18 mm3/sec. (my layer count says 39 as I counted from 1):
[image]

Also of note: the support below curved/rounded edged shapes is excessively jagged.  (This is very pronounced when printing a large object like a helmet, takes a long time to print.)
[image]

Here is the dropbox folder with all the files:
https://www.dropbox.com/sh/cw83vm0th9gkftb/AAB0b7B5N6-wMeO-FxhgRFMra?dl=0

This folder includes:
The config.ini
A spreadsheet with the particular problem layer (other layers also do this), that has all the features broken out, and a second sheet that shows the formulas that were used.
The Gcode (support_test.gcode)
The STL

teppix, 2015-07-30:
1. the middle path that is red is the gap fill.
2. The leftmost path is hanging in the air, and I guess that would make it a bridge? (Not entirely sure about this)

None of these can be set to automatic, so I think that this is expected.
There is already an open issue about adding autospeed for gap fil: https://github.com/alexrj/Slic3r/issues/2976

Paul--W, 2015-07-30:
Thanks for having a look.
1) The middle red path is indicating that it is Infill and I calculate 18.51mm3/sec from the gcode (just a bit over the 18mm3/sec limit).  Here is that line:
G1 X4.588 Y-12.172 E3.82507 F3000.000 ; infill

2) The other paths are indicating perimeter, and are not hanging in air, but are over support material interface (my debugger does not show previous layers but the Preview in Slic3r does):

[image]

Note that my Bridge speed is 40mm/s, and Gap fill at 50mm/s.  What extrusion width are those two used with?  The maximum I can calculate using my perimeter width of .68, .2mm layer height, 50mm/sec, and 1.4 bridge flow ratio (.68 * .20 * 50 * 1.4) = 9.52mm3/sec.

My Speeds:
[image]

My Widths:
[image]

Paul--W, 2015-07-30:
Let me confirm my calculations before looking further.

Paul--W, 2015-07-30:
OK, I've done some further checking, my debugger program was in error.  
In my case the volumetrics are working.  Sorry, and thanks again.

My code missed some changes in feed rate.  The maximum mm3/sec is only 18.17 mm3/sec, so only 0.17 over, within the range of error.

I did learn that the outside line in these supported curved layers is in fact being done as a bridge.  I will have to experiment further to see if the extrusion width in this case is coming out correctly.  I think I should be able to tune the bridging multiplier to make it come out properly - by testing printing some spheres.

I'm not sure if the bridging mm3/sec should be affected by volumetric either, however if volumetric limits are used I think there should be a warning if a setting will cause more than the limit mm3/sec.

closed 2015-07-30 by Paul--W