slic3r/Slic3r #3024: Slicing bug for radar reflector.

AndKe, 2015-07-30:
please see http://madhacker.org/tmp/platonic.stl
The 3d render is perfect, however, the preview have some protruding lines midair, and have a wider top and botton, or "ledge"+bottom in it - than the .STL.

Paul--W, 2015-07-30:
Please check your link:
Oops! That page can’t be found.

AndKe, 2015-07-31:
oops , my bad, please try again :)

Paul--W, 2015-07-31:
I'm looking at this out of curiosity, here's my 2 cents worth:  If you scale it to 500% and look at layer view, you will see that it is in-fact trying to print that little protruding bit in mid-air.  My own g-code debugger confirms this.

From Layers tab:
[image]

BTW, there are some errors in the stl - the normals are reversed.  I corrected them in Meshmixer before doing this test.  It didn't find any other errors in the STL (not that there couldn't be others).

AndKe, 2015-07-31:
the stl is exported from OpenSCAD  - I too did try to fix using netfabb, but it does not help.

teppix, 2015-08-02:
I noticed two interesting things:

1. The problem seems to go away if I disable "detect thin walls", so that might be where the bug is.
2. Since the error only appears on a specific corner, I tried to rotate the object around z, and the problem went away. Rotating 3 degrees removes the problem. Tried 45 degrees, and the problem appears at different locations.

AndKe, 2015-08-03:
Any thought on the fact that the sliced object is wider at the top, than the edges ?

teppix, 2015-08-04:
First of all, I think the stray random extrusions is an obvious bug and needs to be addressed.

Regarding the wide extrusions on top, that also seems to be caused by the detect thin walls feature.

It should also be noted that this model is both very small, and has extremely thin features because of the sharp angles. When accounting for nozzle size (assuming 0.4 mm) you will lose about 10-20% of the corners, calculating in comparison with the entire polygon/triangle.

I'm only speaking for myself here, but as a test case I don't think this is very interesting really, because it's a model that's impossible to print perfectly as it is. You will always have a high margin of error, and it's debatable what exact results you would expect. In some cases you may want to sacrifice/cut off the corners to maintain the shell while in other cases it's desirable to keep the corners at the cost of "breaking" the boundary of the model (extrude all the way to the corner, causing a "fat" corner as you mentioned). Focusing on features that is physically possible to print is simply more interesting in my opinion.

It's of course interesting to get as close to the model as possible, but if you model like this, without concern for the physical limitations of the printer, you can never really know what to expect. I'd bet that if you chamfer the corners so that no feature is below the minimum extrusion width, you'll get near perfect results.

Another question is, while the preview looks a bit odd, how does this look in real life when printed? Perhaps it's the preview that's, while the print comes out acceptable? Not likely perhaps, but it would be interesting to see. After all, it's the printed result that's interesting.

Well, that's just my 2 cents :)

Here's a screenshot with the features smaller than 0.4 mm selected, which is a significant portion of the entire model:
[image]

AndKe, 2015-08-04:
The "edge" phenomenon can be seen even at 600% scale, but yes - I assume the thin "pointy" nature of the item does have something to do with it.
On the contrary, it's very strange to see the "edge" layer be made where other layer's pointy nature is not printed.
It's not a preview-only error, it's in the gcode.

teppix, 2015-08-05:
@AndKe 
After thinking about this a bit, I think you're right that the slicer should do the best possible result given non-optimal input. At the very least for models that are solid. There are obviously some room for improvement here.

I still think that when printing in real life, you should really verify the model for features that will be problematic to print, but I guess that's outside of the scope of issue.

The wide extrusions on the top is probably an approximation done in slic3r, much like what is causing the problems with the gap fill - https://github.com/alexrj/Slic3r/issues/2960. That's just a guess though...