slic3r/Slic3r #3027: Allow changing filename when uploading .gcode to Octoprint

spapadim, 2015-07-31:
This is a feature suggestion related to alexjr/Slic3r#1826: instead of using an automatically determined filename, allow the user to specify it.

Use case argument: when experimenting with a design (e.g., tweaking part wall width in STL to match extrusion width or layer height, tweaking infill or perimeters for strength or flex, etc etc), I like to name the .gcode file to indicate which version of the design/slice it corresponds to, so (i) I'm sure I'm printing the right thing (and not an older slice) and (ii) more rarely, so I can also go back do a repeat trial print without reslicing.  Currently, this is not possible (so in those cases, I still manually export and then upload via octoprint's web UI).

Simple implementation: Just pop up a text entry dialog allowing user to manually modify filename.

Better implementation: First fetch list of files on Octoprint server via it's REST API (GET /api/files/local), and then show a custom dialog with a combo list (allowing re-use of an existing filename or typing a completely new one).  Downside of this option is that it requires parsing JSON (which, as far as I can tell, would be a new dependency for Slic3r, perhaps undesirable?)

If this seems reasonable (and if so, which of the two options would be preferred), I can give a patchset a go...  Thanks!

dcloutier, 2015-09-27:
I had the same problem, I fixed this by adding the following in Expert mode, Print settings/Output options. On the Output filename format: mine now looks:

[input_filename_base]-[timestamp].gcode
which translates to:
mystlfilenameprefix-20150925-135837.gcode

It does not address completely your request but this feature might have gone under the radar.
That way, every new upload to the server has a different name including a datestamp, which is also very handy. That field also accept all of the configuration variables to suit your needs, like layer height, infill density, etc.

Cheers!

alexrj, 2016-12-17:
Done! Slic3r now prompts with the filename and asks whether to start printing.

closed 2016-12-17 by alexrj