slic3r/Slic3r #3029: crash on OS X

ioanniskar, 2015-08-01:
I have Slic3r 1.2.9 on OS X 10.10.4.
Every time I try to export G-code the app crashes.
I found that it works if I select the - default - Print settings, quit slic3r, reopen it, drag my stl and export g-code. The key it to start slic3r with the - default - print settings.
How could I fix this?
Thank you.

lobermann, 2015-08-22:
Same here. Here is the crashlog http://pastebin.com/6AsQna13 .
Works when pushing the code directly to octoprint, but when exporting it crashes. Very very anoying ...

quillford, 2015-09-16:
Brim seems to have a particular issue.

hackermagnet, 2015-09-16:
I don't have brim enabled. I have set it to 0mm.

quillford, 2015-09-16:
OctoPrint upload definitely seems to be way more reliable than exporting gcode.

dionmindhive, 2015-09-30:
Also finding that uploading direct to printer seems to be the only way I can get slic3r to work sometimes. Seemed to be fine exporting to desktop yesterday. Tried a few different versions. Reinstalled. No Joy. 
Bizarrely - after successfully exported some .gcode direct to octoprint I can normally go back o exporting to the desktop. Weird.

Snippet from the crash reports:

Process:               slic3r [2438]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [2438]
User ID:               503

Date/Time:             2015-09-30 13:02:07.947 +1300
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11

Time Awake Since Boot: 38000 seconds
Time Since Wake:       10000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000000b

hackermagnet, 2015-09-30:
After exporting one .gcode file on my Desktop (using the process I described on the first post) I am able to export other .gcode files with my settings. Also I am able to remove the stl or add more and export new .gcode files. When slic3r crashes, the progress bar is loading and it crashes right before the creation of the .gcode file. So the problem might be the creation of the first .gcode file.

dionmindhive, 2015-09-30:
@hackermagnet Yea - I see the same behaviour. Seems like its the last bits of the process that kill it.

hackermagnet, 2015-11-24:
Any way to fix this?
The only way to export the .gcode is this:

1.start slic3r
2.select the -default- print settings
3.quit slic3r
4.restart slic3r
5.import .stl and export .gcode with -default- settings
6.export .gcode with my settings

So slic3r needs to start with the default print settings and perform the first export using these settings.
The problem doesn't exist if I send the .gcode to the printer using Octoprint, so it crashes when creating the .gcode file on my disk. Could it be a permissions issue?

alexrj, 2017-03-17:
We now have a brand new rewritten packaging toolchain for OS X.
You can try the automated builds available at http://dl.slic3r.org/dev/
I'll close this issue as it's very old and very likely to be fixed, but feel free to reopen if you need further assistance.

closed 2017-03-17 by alexrj