slic3r/Slic3r #3030: Free to wrong pool 1b2f3460 not 83555e0 at C:/Slic3r/std/threads.pm line 101

dc42, 2015-08-01:
I just had the above message while trying to export gcode. Slic3r then terminated I am running the win64 build under Windows 7.

dc42, 2015-09-06:
More information:

1. slic3r 1.2.9 crashes on me frequently. I am using the Win64 build under Windows 7.
2. When a load slic3r, my previous Printer and Print settings are always restored, but the Filament setting frequently reverts to "default".
3. Recently I had slic3r crash EVERY TIME I tried to load a part. The sequence was:
(a) Load slic3r
(b) change the filament setting from "default" to one of my saved selections
(c) load the STL file - CRASH.
I eventually managed to slice the part by changing the order to:
(a) Load slic3r
(b) load the STL file
(c) change the filament setting from 'default' to one of my saved selections.

a4jp-com, 2015-09-07:
I also have problem 2, listed above.

milkowski, 2016-05-01:
Manual compilation in Strawberry Perl environment seems to solve that problem on newer Windows. At least 64-bit builds I use have no issues that official binaries exhibit.

alexrj, 2016-12-12:
That should be fixed as of current master thanks to the C++ parallelization I recently pushed. Thank you.

closed 2016-12-12 by alexrj