slic3r/Slic3r #3031: please change custom start gcode order

AndKe, 2015-08-03:
this is the start code we have today.
----------------------
M107 ;fan off
M190 S60 ; set(, and wait for) bed temperature 
M104 S180 ; set hotend temperature (no not wait)
(custom start gcode here)
M109 S180 ; wait for hotend temperature to be reached
--------------------------

the experience is like: 
fan off, set temp, wait for 5 minutes, then set hotend, and wait for additional 3 minutes. 

If you changed the start code order to this;
----------------------
M107 ;fan off
(custom start gcode here)
M190 S60 ; set(, and wait for) bed temperature 
M104 S180 ; set hotend temperature (no not wait)
M109 S180 ; wait for hotend temperature to be reached
--------------------------

Then I could add custom code to achieve this:
fan off, set bed to 35degree and wait, set hotend to 180 (no wait), set bed to 60 (no wait) ...
this would make hotend and bed get warm at about the same time.

whosawhatsis, 2015-08-04:
If you put M109/M190 in your start code, they won't be added automatically, outside of it, so you can put them in the order you want.

AndKe, 2015-08-04:
fantastic, I did not knew the gcode would adapt to what's inside :)

closed 2015-11-04 by alexrj