slic3r/Slic3r #3032: infill

hyperair, 2015-08-04:


hyperair, 2015-08-04:
Example Gcode: https://gist.github.com/54e908ca725fa3cea797
Example STL generated from the following SCAD: cylinder (d = 100, h = 10, $fs = 0.4, $fa = 1);

hyperair, 2015-08-04:
Here's a Python script I used to calculate extrusion width between two lines of G-code: https://gist.github.com/a7e6cd64e1b9f451261f

And here's the snippet I used to calculate the extrusion width with:
```
% layer_height=0.2 extrusion-width << EOF
G1 X48.613 Y0.274 E53.88190 ; infill
G1 X-48.593 Y0.274 E58.38460 ; infill
EOF
0.599998258623

% layer_height=0.4 extrusion-width << EOF
G1 X48.613 Y0.274 E53.88190 ; infill
G1 X-48.593 Y0.274 E58.38460 ; infill
EOF
0.364379680292
```

Paul--W, 2015-08-04:
hyperair,
Thanks for the calculations.
I think I've seen this, and didn't know why, but it worked better to keep the combine infill layers set to 1.

I'm also concerned with pressure loss with a bowden extruder, when printing over sparse infill.  Do you think that could also be at play?

hyperair, 2015-08-05:
> I think I've seen this, and didn't know why, but it worked better to keep the combine infill layers set to 1.

This isn't about combining infill layers though. It's about how rectilinear infill works -- only every alternate layer prints along the same line. Everything else is bridged.

> I'm also concerned with pressure loss with a bowden extruder, when printing over sparse infill. Do you think that could also be at play?

I spent quite some time thinking about that as well, but my conclusion is that Bowden pressure loss ought to happen at the beginning, with a bit of blobbing at the end, but not during the coasting period. The parts that were the most broken in my infill were in the center of the cylinder I printed. The edges actually turned out a bit more solid due to acceleration/jerk slowdown.

Paul--W, 2015-11-21:
Hyperair,
I've done some testing since, with support that uses the same method, every alternate layer prints along the same line, so the same thing may be happening.

I think that <b>since the extrude is not constrained, the extruded shape is mostly circular except where it crosses another line, however the extrude volume is not enough for the circular diameter to reach nozzle diameter as it would for bridging, making the lines too thin and stretched, and easy to break when they cross over another line.</b>  Is this seem right?

This does appear to me to be a real problem, and I've helped others with this on reprap.org.  One way I've attempted to overcome this is by using a larger extrusion width for infill and support.   

For me there is a problem with using the larger extrude width without a corresponding slowdown: it can cause a large change in mm3/sec,  and because of the delay in reaching that increased extrusion rate at the beginning of the first segment (especially if you use a bowden extruder), the start of the infill or support can be starved for some distance.

And I don't now what would be faster and use less filament: printing every line on each layer (can be printed thinner at a faster speed since the bottom is supported), or printing each direction every other layer at the higher mm3/sec that seems to be required - and slower speed?

That's an issue I'm testing with my post-processor - applying a lower acceleration to the very start of a loop on unretract when there is a larger mm3/sec - this gives it time to start properly.  This seems to work well.  However, there also needs to be pressure compensation, bringing the pressure up to what's needed to print at that volume (and reducing it after), I'm testing that too.

The new Max Volumetric Speed option is a good idea, it slows segments down so as not to exceed the max mm3/sec you specify (but only if you have not set those speeds manually).  That's only a partial fix however, as there will still be unavoidably large changes in extrusion rate between outside perimeters/small perimeters (printed thinner and slower), and inside perimeters/infill (printed wider at higher mm3/sec) for instance.  

Since this is all about mm3/sec, really the speed, extrusion width, and acceleration settings for each kind of printing should be in the same place in Slic3r, and should show the mm3/sec that will occur.  Everyone needs to know what mm3/sec that their printer can handle.  I'm working on a test generator for this and other cases.

vlast3k, 2015-11-21:
+1
I have a similar problem. I am printing 1.3 mm wide, single line walls, 0.2 mm layer height at 40 mm/s with a 0.5 nozzle.
each layer starts with ~ 7mm of thin line until the pressure ramps up.
the only way to make it print well was to  create a vertical hole in the wall and have a 2 mm edge to the inside and slower retract speed for this segment. I am not sure if this can be solved any other way than empirically. But it would be nice. Also tried decreasing the acceleration to 100 mm/s2, but it did not help a lot

hyperair, 2015-11-23:
On Sat, Nov 21, 2015 at 01:06:24PM -0800, Paul--W wrote:
> Hyperair,
> I've done some testing since, with support that uses the same method, every
> alternate layer prints along the same line, so the same thing may be
> happening.
>
> I think that <b>since the [...]

Yes, that is right.

> This does appear to me to be a real problem, and I've helped others with this
> on reprap.org. One way I've attempted to overcome this is by using a larger
> extrusion width for infill and support.
>
> For [...]

Correction: it causes a large change in mm³/mm. If you use Slic3r's autospeed,
then Slic3r will try to keep mm³/s constant, resulting in a slower mm/s for
those sections. Because mm³/s is kept constant, you don't get that delay in
extrusion that causees underextrusion at the beginning of infill lines.

In a nutshell, use autospeed if you have issues with large mm³/s steps.

> [...]
>
> The new Max Volumetric Speed option is a good idea, it slows segments down so
> as not to exceed the max mm3/sec you specify (but only if you have not set
> those speeds manually). That's only a [...]

Maximum volumetric speed and maximum planner speed form the basis for Slic3r's
autospeed. Specifically, Slic3r looks for the thinnest lines (smallest mm³/mm),
and prints those at the highest allowable speed (maximum planner speed) while
obeying maximum volumetric speed. This gives the target volumetric speed Slic3r
will try to maintain throughout the print, and Slic3r figures out the print
speeds for each extrusion type based on that.

I don't think everyone needs to know their maximum volumetric speed. My print
speeds are usually slow enough that it doesn't hit the threshold (24mm³/s or so)
despite doing some print moves at 200mm/s.

-- 
Kind regards,
Loong Jin

ghuser2, 2015-11-27:
Turning on maximum volumetric speed, and setttin it to 5, solved some big problems for me. Probably because of a PTFE liner limiting the transfer of heat.

hyperair, 2015-11-27:
On Thu, Nov 26, 2015 at 08:03:31PM -0800, ghuser2 wrote:
> Turning on maximum volumetric speed, and setttin it to 5, solved some big
> problems for me. Probably because of a PTFE liner limiting the transfer of
> heat.

Ah, that makes sense then. A top speed of 5mm³/s is really poor performance
though.

-- 
Kind regards,
Loong Jin