slic3r/Slic3r #3034: I couldn't compile the latest Slicer development version on Windows 10 (64-bit)

a4jp-com, 2015-08-07:
I couldn't compile the latest Slicer development version on Windows 10 (64-bit)

Any advice is greatly appreciated.

I loaded the latest version of Strawberry Perl (32-bit).

-----------------------------------------------------
c:\slic3r-4>perl -v

This is perl 5, version 22, subversion 0 (v5.22.0) built for MSWin32-x86-multi-thread-64int

Copyright 1987-2015, Larry Wall

Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.

[image]

a4jp-com, 2015-08-08:
Any ideas anyone?

llluis, 2015-08-10:
The problem is with the module ExtUtils::CppGuess (which is a prerequisite of other modules).
I had the same problem and is a known bug with this module: https://github.com/tsee/extutils-cppguess/issues/9

You will have to make the change in the source indicated in that issue and install the module manually.

Or you can download the corrected package here: https://www.dropbox.com/s/hpzipgbsm7ngaf1/ExtUtils-CppGuess-0.09.tar.gz?dl=0

And run:

cpanm /path/to/the/file

After that run again the Build.PL

a4jp-com, 2015-08-10:
Thank you soooooo much @llluis.

Do you mean path to the Build.PL file?

a4jp-com, 2015-08-10:
[image]

I guess the path is to the files you posted. I'll try again.

a4jp-com, 2015-08-10:
[image]

I'm not sure which folder or file to point to.

a4jp-com, 2015-08-10:
I changed cpanm to cpan and it tried to install but failed.

[image]

a4jp-com, 2015-08-10:
It is compiling now but gets stuck on Building and testing Wx-0.9927 ... when making the gui file.

[image]

llluis, 2015-08-11:
You had to point to the file I sent.
'''cpanm ExtUtils-CppGuess-0.09.tar.gz'''
But as it's working now I believe you figured this out.

For the Wx, do you see a window with the title "X" during compilation? It take a long time to compile this module.

a4jp-com, 2015-11-08:
I don't have this problem as I'm using version 1.2.10-dev now.

closed 2015-11-08 by a4jp-com