slic3r/Slic3r #3036: GUI Not working(but I can run slic3r.pl just fine)

matmo-xx, 2015-08-10:
So I have a computer running Ubuntu, and I can run slic3r in terminal, but I don't get the GUI. 

Input: ./slic3r.pl --gui

Result: Can't locate OpenGL.pm in @INC (you may need to install the OpenGL module) (@INC contains: /home/owner/Desktop/printing/Slic3r/lib /etc/perl /usr/local/lib/perl/5.18.2 /usr/local/share/perl/5.18.2 /usr/lib/perl5 /usr/share/perl5 /usr/lib/perl/5.18 /usr/share/perl/5.18 /usr/local/lib/site_perl .) at /home/owner/Desktop/printing/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
BEGIN failed--compilation aborted at /home/owner/Desktop/printing/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
Compilation failed in require at /home/owner/Desktop/printing/Slic3r/lib/Slic3r/GUI.pm line 16.
BEGIN failed--compilation aborted at /home/owner/Desktop/printing/Slic3r/lib/Slic3r/GUI.pm line 16.
Compilation failed in require at (eval 70) line 1.

Can someone please help me?

llluis, 2015-08-10:
perl Build.PL --gui

llluis, 2015-08-10:
perl Build.PL --gui

matmo-xx, 2015-08-10:
I already ran that command. This time, it's telling me that

Configuring Wx-GLCanvas-0.09 ... OK
Building and testing Wx-GLCanvas-0.09 ... FAIL
! Installing Wx::GLCanvas failed. See /home/owner/.cpanm/work/1439222724.19312/build.log for details. Retry with --force to force install it.
Don't worry, this module is optional.

llluis, 2015-08-10:
Look at the indicated build.log and see if you have the following error:

```
Test Summary Report
-------------------
t/04_userdata.t     (Wstat: 0 Tests: 65 Failed: 0)
  TODO passed:   25, 33, 41, 57
t/14_eh_die.t       (Wstat: 0 Tests: 5 Failed: 0)
  Parse errors: Bad plan.  You planned 6 tests but ran 5.
Files=23, Tests=791, 1925 wallclock secs ( 0.16 usr +  0.17 sys =  0.33 CPU)
Result: FAIL
Failed 1/23 test programs. 0/791 subtests failed.
dmake.exe:  Error code 255, while making 'test_dynamic'
```

llluis, 2015-08-10:
If not, send the build.log

matmo-xx, 2015-08-10:
owner@owner-Satori:~/Desktop/printing/Slic3r$ perl Build.PL --gui
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
App::cpanminus is up to date. (1.7039)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Class::Accessor is up to date. (0.34)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Growl::GNTP is up to date. (0.20)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
LWP::UserAgent is up to date. (6.13)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Net::Bonjour is up to date. (0.96)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.6704.tar.gz ... OK
Configuring OpenGL-0.6704 ... N/A
! Configure failed for OpenGL-0.6704. See /home/owner/.cpanm/work/1439222713.19283/build.log for details.
Don't worry, this module is optional.
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Socket is up to date. (2.020)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Wx is up to date. (0.9927)
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
Configuring Wx-GLCanvas-0.09 ... OK
Building and testing Wx-GLCanvas-0.09 ... FAIL
! Installing Wx::GLCanvas failed. See /home/owner/.cpanm/work/1439222724.19312/build.log for details. Retry with --force to force install it.
Don't worry, this module is optional.

llluis, 2015-08-10:
Need the /home/owner/.cpanm/work/1439222724.19312/build.log

llluis, 2015-08-10:
And /home/owner/.cpanm/work/1439222713.19283/build.log

matmo-xx, 2015-08-10:
How do I access the build logs?

llluis, 2015-08-10:
I've pasted the path to them.
/home/owner/.cpanm/work/1439222724.19312/build.log
/home/owner/.cpanm/work/1439222713.19283/build.log

Copy and paste them here.

matmo-xx, 2015-08-10:
/home/owner/.cpanm/work/1439222724.19312/build.log:

cpanm (App::cpanminus) 1.7039 on perl 5.018002 built for x86_64-linux-gnu-thread-multi
Work directory is /home/owner/.cpanm/work/1439222724.19312
You have make /usr/bin/make
You have LWP 6.13
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip

!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Searching Wx::GLCanvas (0) on cpanmetadb ...
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz
-> OK
Unpacking Wx-GLCanvas-0.09.tar.gz
Entering Wx-GLCanvas-0.09
Checking configure dependencies from META.yml
Checking if you have Wx::build::MakeMaker 0.16 ... Yes (0.28)
Configuring Wx-GLCanvas-0.09
Running Makefile.PL
Checking if your kit is complete...
Looks good
'WX' is not a known MakeMaker parameter name.
Generating a Unix-style Makefile
Writing Makefile for Wx::GLCanvas
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.04)
Checking if you have Wx 0.57 ... Yes (0.9927)
Checking if you have Wx::build::MakeMaker 0.16 ... Yes (0.28)
Building and testing Wx-GLCanvas-0.09
cp lib/Wx/DemoModules/wxGLCanvas.pm blib/lib/Wx/DemoModules/wxGLCanvas.pm
cp lib/Wx/GLCanvas.pm blib/lib/Wx/GLCanvas.pm
"/usr/bin/perl" -S wxperl_overload cpp/ovl_const.cpp cpp/ovl_const.h overload.lst
Writing 'cpp/ovl_const.h'.
Writing 'cpp/ovl_const.cpp'.
touch wxt_overload
Running Mkbootstrap for Wx::GLCanvas ()
chmod 644 "GLCanvas.bs"
"/usr/bin/perl" "/usr/local/share/perl/5.18.2/ExtUtils/xsubpp"  -noprototypes  -typemap "/usr/share/perl/5.18/ExtUtils/typemap" -typemap "/usr/local/lib/perl/5.18.2/Wx/typemap" -typemap "typemap"  GLCanvas.xs > GLCanvas.xsc && mv GLCanvas.xsc GLCanvas.c
g++ -pthread   -c  -I. -I/usr/local/lib/perl/5.18.2/Wx -I/usr/lib/x86_64-linux-gnu/wx/include/gtk2-unicode-release-2.8 -I/usr/include/wx-2.8  -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.09\" -DXS_VERSION=\"0.09\" -fPIC "-I/usr/lib/perl/5.18/CORE"  -DWXPL_EXT -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES -D__WXGTK__   GLCanvas.c
In file included from /usr/include/wx-2.8/wx/glcanvas.h:54:0,
                 from GLCanvas.xs:27:
/usr/include/wx-2.8/wx/gtk/glcanvas.h:19:19: fatal error: GL/gl.h: No such file or directory
 #include <GL/gl.h>
                   ^
compilation terminated.
make: *** [GLCanvas.o] Error 1
-> FAIL Installing Wx::GLCanvas failed. See /home/owner/.cpanm/work/1439222724.19312/build.log for details. Retry with --force to force install it.
(END)

matmo-xx, 2015-08-10:
/home/owner/.cpanm/work/1439222713.19283/build.log:

cpanm (App::cpanminus) 1.7039 on perl 5.018002 built for x86_64-linux-gnu-thread-multi
Work directory is /home/owner/.cpanm/work/1439222713.19283
You have make /usr/bin/make
You have LWP 6.13
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
!
! Can't write to /usr/local/share/perl/5.18.2 and /usr/local/bin: Installing modules to /home/owner/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.18.2 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
Searching OpenGL (0) on cpanmetadb ...
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.6704.tar.gz
-> OK
Unpacking OpenGL-0.6704.tar.gz
Entering OpenGL-0.6704
Checking configure dependencies from META.json
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.04)
Configuring OpenGL-0.6704
Running Makefile.PL
glversion.c:8:25: fatal error: GL/freeglut.h: No such file or directory
 #include <GL/freeglut.h>
                         ^
compilation terminated.
make: *** [glversion.o] Error 1
-> N/A
-> FAIL Configure failed for OpenGL-0.6704. See /home/owner/.cpanm/work/1439222713.19283/build.log for details.

matmo-xx, 2015-08-10:
Btw thank you so much for your help

llluis, 2015-08-10:
Are you using Ubuntu? Any particular reason for not running this with sudo?

I don't have much experience with Perl + Slic3r + Linux, but your logfile mentions

 ```glversion.c:8:25: fatal error: GL/freeglut.h: No such file or directory``` 

and Slic3r instructions for Linux states 

```Note for Linux users: you need to install the freeglut package provided by your distro.```

Have you installed it?

Also, did you execute all the steps described in https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux ?

llluis, 2015-08-10:
This seems to be an old problem, and as I mentioned, related to pre-requisites: https://github.com/alexrj/Slic3r/issues/2674

matmo-xx, 2015-08-12:
I tried to install freeglut, but I got this error message, and the GUI still doesn't work

owner@owner-Satori:~$ sudo apt-get install freeglut3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 unity-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

yushuiqiang, 2016-05-05:
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ......... ok
t/02_inheritance.t .. ok
t/03_threads.t ...... ok
t/04_taghandlers.t .. ok
t/08_ovl_func.t ..... ok
All tests successful.
Files=5, Tests=31,  2 wallclock secs ( 0.05 usr +  0.02 sys =  0.06 CPU)
Result: PASS
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- DND.
bs ..\..\blib\arch\auto\Wx\DND\DND.bs 644
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ......... ok
t/02_inheritance.t .. ok
t/03_pldatabject.t .. ok
t/04_threads.t ...... ok
All tests successful.

Test Summary Report
-------------------
t/03_pldatabject.t (Wstat: 0 Tests: 9 Failed: 0)
  TODO passed:   7, 9
Files=4, Tests=15,  1 wallclock secs ( 0.05 usr +  0.02 sys =  0.06 CPU)
Result: PASS
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- DocV
iew.bs ..\..\blib\arch\auto\Wx\DocView\DocView.bs 644
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ......... ok
t/02_inheritance.t .. ok
t/03_threads.t ...... ok
t/04_cmdproc.t ...... ok
All tests successful.
Files=4, Tests=40,  1 wallclock secs ( 0.05 usr +  0.02 sys =  0.06 CPU)
Result: PASS
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- Medi
a.bs ..\..\blib\arch\auto\Wx\Media\Media.bs 644
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ......... ok
t/02_inheritance.t .. ok
All tests successful.
Files=2, Tests=4,  0 wallclock secs ( 0.00 usr +  0.03 sys =  0.03 CPU)
Result: PASS
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- Cale
ndar.bs ..\..\blib\arch\auto\Wx\Calendar\Calendar.bs 644
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ......... ok
t/02_inheritance.t .. ok
t/03_threads.t ...... ok
All tests successful.
Files=3, Tests=10,  1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)
Result: PASS
"C:\Strawberry\perl\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- Date
Time.bs ..\..\blib\arch\auto\Wx\DateTime\DateTime.bs 644
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-
e" "undef *Test::Harness::Switches; test_harness(0, '..\..\blib\lib', '..\..\bli
b\arch')" t/*.t
t/01_load.t ..... ok
t/02_api.t ...... ok
t/03_threads.t .. ok
All tests successful.
[... 205 lines cut ...]
.t line 8.
t/18_unicode.t ........ ok
t/19_overlay.t ........ ok
t/20_ctx_threads.t .... ok
t/21_locale.t ......... ok
t/22_interface.t ...... ok
t/zz_distrib.t ........ skipped: Author tests

Test Summary Report
-------------------
t/04_userdata.t     (Wstat: 0 Tests: 65 Failed: 0)
  TODO passed:   25, 33, 41, 57
t/14_eh_die.t       (Wstat: 0 Tests: 5 Failed: 0)
  Parse errors: Bad plan.  You planned 6 tests but ran 5.
Files=23, Tests=791, 136 wallclock secs ( 0.19 usr +  0.03 sys =  0.22 CPU)
Result: FAIL
Failed 1/23 test programs. 0/791 subtests failed.
dmake.exe:  Error code 255, while making 'test_dynamic'
  MDOOTSON/Wx-0.9928.tar.gz
  C:\STRAWB~1\c\bin\dmake.exe test -- NOT OK
//hint// to see the cpan-testers results for installing this module, try:
  reports MDOOTSON/Wx-0.9928.tar.gz
Stopping: 'install' failed for 'Wx'.

C:\Users\Administrator\Documents>

nicksears, 2016-05-26:
Anyone get anywhere with this? I've compiled from source before but I think I used citrus perl.
I got restless and didn't want to wait for the dreaded gap fill fix any longer...

I can compile with Build.PL, but not the GUI. I notice the following errors when compiling (not GUI)

Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
Fatal error: can't create buildtmp\XS.exp: Permission denied
g++: error: buildtmp\XS.exp: No such file or directory

This seems to be a success, but I dont know how to use this version (compiled from source) with repetier, since it looks for an exe. I've seen somewhere that it's possible by passing a batch file with parameters for Repetier, but I don't want to mess with that.

When I compile with --gui I get these errors when it gets to Wx:
When it gets to "Building and testing Wx-0.9928 ..."
It eventually opens a window labeled "X"and hangs (says "not responding" I've let it go for hours)
I can cancel this one window, but it says:
"! Installing Wx failed. See C:\Users\nick\.cpanm\work\1464232311.4440\build.log for details. Retry with --force to force install it."

I've read to try and install/update wxwidgets and Alien::wxWidgets, and then try to update Wx, but dont know why that would work.

lordofhyphens, 2016-05-26:
There are apparently libs and tools with strawberry perl you can use to
pack an exe (perl2exe ?) with the interpreter baked in. I know python has a
similar capability. I just don't dev on Windows at all. I have a discussion
issue at #3336 for this.

jharmn, 2016-12-17:
FYI I discovered an Ubuntu/Mint fix for this in https://forum.lulzbot.com/viewtopic.php?t=5437

  sudo apt install wx-common libopengl-perl libwx-glcanvas-perl

Mine is running fine now.

lordofhyphens, 2017-02-22:
@nicksears https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev for set of dependencies for windows. 

For linux, see one of the various workarounds (on my system, straight builds work too). 
Closing.

closed 2017-02-22 by lordofhyphens