slic3r/Slic3r #3038: Verbose output doubling the ";"

llluis, 2015-08-10:
The current version outputs double " ; " in some comments when verbose is enabled:

```
G1 X21.881 Y15.630 E0.02668 ;  ; skirt
G1 X-0.695 Y3.632 E0.00736 ;  ; perimeter
G1 X2.856 Y-18.798 E0.35908 ;  ; infill
```

This seems to be caused by GCode (line 579):

```
std::string comment = this->config.gcode_comments ? (" ; " + description) : "";
```

and GCoderWriter (line 9):
```
#define COMMENT(comment) if (this->config.gcode_comments && !comment.empty()) gcode << " ; " << comment;
```

lordofhyphens, 2015-08-10:
Looks like submitting a patch would be easy.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

llluis, 2015-08-10:
Done.
:)

closed 2015-11-04 by alexrj