slic3r/Slic3r #3040: Weird moves

hippich, 2015-08-11:
Using -dev version, I noticed couple things, see video and comments below:

https://youtu.be/WesVmcfFnaA
[image]

- when I am printing two copies, generated code include weird move close to x:0 y:0 area. Not sure why it doing it. May be it is a feature? It actually cause pretty bad blobs where extruder enters part after this move. This is due plastic oozing while it does this move.

- printing perimeters also weird. it jumps from one hole to another, just to jump back and finish perimeter there. this slows down print over all and also causes extra small blobs, although not as bed as with weird move above.

Any ideas on what to adjust to prevent that? I am attaching generated g-code and stl model I tried to print.

STL: https://github.com/hippich/stl/blob/master/iris/tarot_adapter/iris5_2mm.stl
GCode: https://github.com/hippich/stl/blob/master/iris/tarot_adapter/iris5_2mm.gcode

chorca, 2015-08-13:
Just as a note, I experience this as well with the latest version from Git. Upon finishing each piece, the head drives back to around origin, then springs out to the next piece. I've also noticed the perimeter issue on one of my latest prints, where perimeters are printed at different times on the same piece.

chorca, 2015-08-16:
Just a quick note on this. I noticed a lot of issues printing with the latest dev version (probably because it's still in dev!) so just wanted to let the author know about the bug..
I also see odd excessive, unnecessary moves around the part while printing, and also there seems to be an issue where a wipe operation causes a very slow retract to occur while moving at slow speed.

Looks like the last good commit was 3e739b87da9f49b61a6dfc9aa99a06ba28ddfe39, after that, the gcode generated is about 2x the size with a lot of odd moves and such. I'll stick on this commit and test the next one.

roboprint, 2015-08-23:
This weird moves are exist in slicer in older 1.2.x versions too.

hippich, 2015-08-25:
@roboprint, i am using 1.2.9 version from website right now and don't experience sudden jumps between perimeters and weird move to area of X0 Y0 between part copies.

alexrj, 2015-12-19:
Hello, this should be fixed as of 667a121ddbbe9732c3831166792128a901c01315

hippich, 2015-12-19:
Confirmed as fixed

closed 2015-12-19 by hippich