slic3r/Slic3r #3045: [1.2.9-dev] Slic3r autospeed causing negative F values to be output in gcode

lordofhyphens, 2015-08-19:
On the provided STL, setting autospeed for perimeters causes negative speed values to be used in output gcode.

I can confirm it doesn't have this behavior on the 5mm calibration step cube on youmagine/thingiverse

Test stl and config:
http://www.2shared.com/file/ckhcBgcM/negative-accel.html

W3DRK, 2015-12-16:
I have also witnessed this using Slic3r built from the current source.  It has occurred on every STL I've tried slicing so far, and causes my print controller to freeze.

Here's an example snippet of offending GCode:

G1 Z0.450 F7800.000
G1 X129.694 Y121.278 F7800.000
G1 Z0.250 F7800.000
G1 E4.00000 F6000.00000
G1 F-360.595
G1 X106.300 Y121.278 E5.15010
G1 X106.296 Y120.642 E5.18137
G1 X109.185 Y120.639 E5.32340
G1 X109.730 Y120.572 E5.35041

koghe, 2015-12-16:
I have the same issue https://github.com/alexrj/Slic3r/issues/3108#issuecomment-156864235
was some days ago, not tried with the current git because a don't want to clog my nozzle again...

[image]

remove autospeed slicing is correct 
[image]

ini : https://www.dropbox.com/s/zeu3vmjcnhtqixz/config-autospeed-bug.ini?dl=0
stl : https://www.dropbox.com/s/eb7a36wj5c33p9g/coolingFan-new.stl?dl=0
gcode : https://www.dropbox.com/s/ut30b0o0n4tvznh/coolingFan-new.gcode?dl=0

lordofhyphens, 2016-01-18:
I think part of it is the model itself, some ratio of infill to perimeters. I think you can make it happen more by setting perimeters to be some multiple of extrusion of the perimeters (say 0.5mm perimeter and 1-2mm infill).

alexrj, 2016-03-18:
I tested @lordofhyphens's STL and config on 1.2.10-dev (stable branch) and I couldn't find any occurrence of the "F-" string in the G-code. Will test on master.

alexrj, 2016-03-18:
I also tested @kino3d's file with current stable branch, with same result. Maybe this bug was fixed in 19cb1dee63ced68ee160ccca8504a5599ff85147
@lordofhyphens, @kino3d, @W3DRK, can you confirm this is fixed on your side as well?

kino3d, 2016-03-18:
Hi Alex 
yes,my file is fixed with the same config and new one :+1:

alexrj, 2016-03-18:
Thank you for confirming that @kino3d.

kino3d, 2016-03-20:
Hi @alexrj with the last commits i have again negative values

[image]

stl https://www.dropbox.com/s/y7vlpj6u7dftvqh/desk-Circle.stl?dl=0
gcode https://www.dropbox.com/s/ivdj98clovbezjh/desk-Circle.gcode?dl=0
config https://www.dropbox.com/s/zx1j8aro4tqxfko/autospeed-negative-config.ini?dl=0

lordofhyphens, 2016-03-20:
Not that there is anything wrong with Dropbox, but you can upload ZIP files
to GitHub now :). Just FYI.

kino3d, 2016-03-21:
Yes i'm forgeting every time :) 
next time will do

lordofhyphens, 2016-05-27:
@kino3d could you try my PR and see if it helps for you and doesn't cause something else to go weird?

kino3d, 2016-05-27:
@lordofhyphens thanks to take care of this :+1: 
from the gcode.ws seem ok, will test with new models
thanks

Model size is: 48.05x62.36x34.75mm
Total filament used: 10507.88mm
Total filament weight used: 26.28grams
Estimated print time: 35:57:12
Estimated layer height: 0.25mm
Layer count: 142printed, 143visited
Time cost: 35.95
Filament cost: 1.31

lordofhyphens, 2016-05-27:
Yeah it basically is the same thing as running a pass over the gcode and
deleting all negative feedrates, so there are probably going to be side
effects.

kino3d, 2016-05-27:
yes, just trying the same conf with a new model and the result is bad

[image]

alexrj, 2016-07-10:
@kino3d, I tested your desk-Circle file with current master and I cannot reproduce the issue. Can you confirm it's fixed?

alexrj, 2016-07-10:
Also, @kino3d, your G-code file is not available anymore.
I need to reproduce the issue before taking any action.

alexrj, 2016-07-10:
I just pushed a fix for #3220 so I think this issue might be fixed as well. Let me know.

kino3d, 2016-07-10:
@alexrj  
here is it model, i lost the first gcode on cleaning  dropbox, this one is from the Lord patch,
[deskcircle.zip](https://github.com/alexrj/Slic3r/files/356019/deskcircle.zip)

I cannot confirm sorry, i have the same bug with a completely new config 
let me know how i can help you on debug 
linux ubuntu 15.04
perl 5, version 20, subversion 2 (v5.20.2) built for x86_64-linux-gnu-thread-multi
here the config and gcode, model is always the same 
[autospeedbug.zip](https://github.com/alexrj/Slic3r/files/356025/autospeedbug.zip)

kino3d, 2016-07-10:
ok let me try last fix

kino3d, 2016-07-10:
nope sorry 

[desk-Circlefix3.gcode.zip](https://github.com/alexrj/Slic3r/files/356070/desk-Circlefix3.gcode.zip)

alexrj, 2017-03-14:
@kino3d, the latest G-code files you attached don't contain any negative F value. Since this is an old report I'm closing this, but feel free to comment and reopen if you think the bug persists.

closed 2017-03-14 by alexrj