slic3r/Slic3r #3046: Vase mode won't start if support is enforced but turned off

VikOlliver, 2015-08-20:
The confusing message "Spiral vase mode is not compatible with support material" appears if a spiral vase mode Gcode output is attempted and support material is turned off. This happens when support is enforced for the first layer - although it won't actually be generated because support is turned off...

Took me a while to figure out what was going on there!

a4jp-com, 2015-08-20:
Nice find. Got a sample file and config file for testing?

VikOlliver, 2015-08-21:
It happens with any STL file. Not sure what a config file is - ~/.Slic3r contains a whole heap of .ini files. It might just be easier to start with any vase config that works for you, go to Print Settings/Support Material and make sure "Generate support material" is unticked. Then set "Enforce Support for the first:" to 1.

I'm fairly sure the problem lies here:
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Config.pm#L300

where there is an unnecessary check of support_material_enforce_layers.

Vik :v)

lordofhyphens, 2015-08-21:
Export your config from the file menu.

Or make the fix, test it, and submit a pull request :)

a4jp-com, 2015-08-21:
The files are needed just to check if the problem is just on one PC or every PC.

Also the wall thickness can make a difference depending on the model.

Do you have a link to a model with this problem? I don't have any vases to test and see if I have the same problem. All of my files have been coming out pretty well though.

File > Export config bundle...

VikOlliver, 2015-08-22:
It is not possible to export the config. As I pointed out earlier, there is suspicious logic in Config.pm#L300. As this is called in any attempt to export the config, attempting to export the config will also reproduce the error!

a4jp-com, 2015-08-22:
Can you take a screen shot of the settings then?

thebobgeezer, 2015-10-14:
Thanks for the help. I was lost.  On the repetier host for print-rite, disabling support and setting enforce to 0 was good.  Happy building all.
B0B

hyperair, 2016-01-17:
The tooltip message for "support_material_enforce_layers" suggests that it overrides the "support_material" option. Therefore the bug is not in the check that is done, but is more of a GUI bug where the option is grayed out when you uncheck "Generate support material".

    def->tooltip = "Generate support material for the specified number of layers counting from bottom, regardless of whether normal support material is enabled or not and regardless of any angle threshold. This is useful for getting more adhesion of objects having a very thin or poor footprint on the build plate.";

VikOlliver, 2016-01-17:
I thought that, but I'm using v1.2.9 and I don't get support generated unless support is turned on, whatever is in that tick box. Am I just strangely unlucky?

hyperair, 2016-01-21:
You're right. So `support_material` turns off `support_material_enforce_layers`, but the tooltip for `support_material_enforce_layers` says that shouldn't happen. Additionally, the vase mode does the `support_material || support_material_enforce_layers > 0` check.

So, which behaviour is correct?

PR #3209 (2016-03-13): Don't toggle support_material_enforce_layers field
support_material_enforce_layers works independently of the support_material ||
raft options, so we should not disable the field when support material
generation is disabled.

Fixes: #3046
commit a9c5874d (2016-03-13): Don't toggle support_material_enforce_layers field

support_material_enforce_layers works independently of the support_material ||
raft options, so we should not disable the field when support material
generation is disabled.

Fixes: #3046
commit c903f839 (2016-03-14): Don't toggle support_material_enforce_layers field

support_material_enforce_layers works independently of the support_material ||
raft options, so we should not disable the field when support material
generation is disabled.

Fixes: #3046
closed 2016-03-13 by alexrj