slic3r/Slic3r #3047: Feature request: choose face to be the base

MoonshineSG, 2015-08-20:
Allow the user to chose which face to be the base of the model and autorotate model would be a time saving feature.

And please write X Y Z on the axes....

lordofhyphens, 2016-05-29:
X/Y/Z are now colored and have proper notes on them.

While I think a "lay flat" option would be convenient in the plater, the effort to implement your request is substantial enough that it probably won't get done by the people currently working on code. Good news is that it's all in Perl, so it's fairly straightforward to bodge something together that mostly works. 

If you want to get this in, probably best to get it done yourself or bribe another coder to work on it and submit a PR.

MoonshineSG, 2016-05-29:
Thanks for the update. I've since then, moved away from Slic3r

lordofhyphens, 2018-05-04:
@MoonshineSG does the following statement describe your wish?

---

**As** a user, I want to be able to rotate models so that a given face lay flat against the build plate, **so** that I can manage plating difficult 3D models in Slic3r and not have to work out the rotation and translation math by hand.

---

This feature request was selected to be worked on during Google Summer of Code 2018, its process will be tracked in #4397

MoonshineSG, 2018-05-05:
I am no longer using Slic3r, but yes, that was what I was looking for.  A simple way to rotate and "drop" a model.