slic3r/Slic3r #3049: Feature Request: New raft calculation algorithm

printerguy, 2015-08-21:
Hello,

First thanks for this great software!

I have an issue with the current raft algorithm that can be improved. This is the situation:

[image]

The situation here is the raft only encapsulate the first layer of the actual object not the support. This causes difficulties to stick the support to the bed.

The other thing is that if there is a wider layers than the first layer, raft can not help to protect from the heat that comes from the bed.

I think the raft algorithm can use infilled skirt and the problem would be solved.

What do you think?

koghe, 2015-08-24:
Hi
which version of slic3r, system, config? 

here is the git version, seem it work

[image]

printerguy, 2015-08-24:
I use the master branch that was packaged by me and the latest released version (1.2.9) but the result is same.

The example you showed is `brim` not `raft`. Most of the time `brim` doesn't solve the problems I described in the issue.

Here is the screenshot of the first layer:

[image]

As you can see there is no raft layers before the first support layers.

printerguy, 2015-08-24:
I use the master branch that was packaged by me and the latest released version (1.2.9) but the result is same.

The example you showed is `brim` not `raft`. Most of the time `brim` doesn't solve the problems I described in the issue.

Here is the screenshot of the first layer:

[image]
[image]

Actually the first layer of the support is not a raft layer

koghe, 2015-08-24:
understand what you mean now, never had this type of problem

a4jp-com, 2015-08-24:
How did you make the support? Got a sample model?

printerguy, 2015-08-24:
Oh sorry. Here is the example: https://www.dropbox.com/s/sxytcz5mi68nrdt/liberty.stl?dl=0

The issue occurs when there is support material outside of the first layer of the actual object.

a4jp-com, 2015-08-24:
If I click on the model I can see the support material options but want to know which one to select to test.

printerguy, 2015-08-24:
Here is the more detailed screenshot of the **second layer** with `raft_layers = 2`:

[image]

Here is my config:

```
# generated by Slic3r 1.2.9 on Mon Aug 24 19:02:55 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,200x0,200x200,0x200
bed_temperature = 120
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 80
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 0
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 70%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_diameter = 1.75
fill_angle = 45
fill_density = 20%
fill_pattern = line
first_layer_acceleration = 0
first_layer_bed_temperature = 120
first_layer_extrusion_width = 200%
first_layer_height = 0.3
first_layer_speed = 40%
first_layer_temperature = 270
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 120
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 80
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 2
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 0.5
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 80
seam_position = aligned
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 30
small_perimeter_speed = 50
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 120
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 1
support_material_angle = 40
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 1
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 2
support_material_speed = 120
support_material_threshold = 20
temperature = 250
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 160
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0
```

printerguy, 2015-09-06:
Here is how skeinforge raft works: http://fabmetheus.crsndoo.com/wiki/index.php/Skeinforge_Raft

koghe, 2015-09-07:
slic3r raft work in a different way https://github.com/alexrj/Slic3r/issues/1323#issuecomment-21620753
having more control like skeinforge will be really good

lordofhyphens, 2016-06-28:
The first layer of the support is the same as raft.

bubnikv, 2017-01-31:
The current prusa3d master works the following way:

Supports with a raft
[image]

Supports without a raft
[image]

It puts the raft below the support and it inflates the 1st layer. One may argue that it is a waste of filament, but it is safe and stable.