slic3r/Slic3r #3053: Feature request: retract on first layer

kloppertje, 2015-08-30:
Using Slic3r 1.2.9: In order to reduce the number of retracts, I enabled the option "Only retract when crossing perimeters". However, I'd like to see an option to disable this function for the first layer, so that no ooze is visible in the bottom of a printed object. Would this be possible? If required, I can try to make a photo of the issue.

mrvn, 2015-09-26:
See https://github.com/alexrj/Slic3r/issues/3068 for a more general feature request for more retract options. I noticed the same problem.

alexrj, 2015-12-19:
Since Slic3r now ignores that option for the top external layers, maybe we could do the same for the bottom, without even adding an option...

xoan, 2015-12-19:
Nice solution :)

kloppertje, 2015-12-21:
Seems like a perfect solution to me.

alexrj, 2016-12-16:
Done!

closed 2016-12-16 by alexrj