slic3r/Slic3r #3057: Feature request: lift Z on retraction, except on first N layers

dc42, 2015-09-06:
Please can the "Lift Z" option in Printer settings -> Extruder 1 have a "except on first N layers" option. I generally find "Lift Z" works well, however on prints with sharp pointy corners it tends to cause that part of the print to detach from the bed if a lift occurs at that point in the first layer or two.

a4jp-com, 2015-09-07:
Maybe an option like that would help but also changing the heat settings is another way to fix this. Can you post a picture of one of your printed models with the problem? What type of print bed do you have? Also are you using a skirt and raft?

luke321, 2015-09-09:
I am currently working on a Gcode Post Processor which eliminates z Lift for a certain number of layers and for short travel distances.
This is definitely a nice feature to reduce print time and increase print quality!

a4jp-com, 2015-09-09:
Cool. I always went with the heat settings to fix problems like this but anything that makes printing easier is greatly welcomed ^^

Lenbok, 2015-09-11:
This would also be addressed by #1316  (which discusses e.g. auto adding "mouse ears" on sharp corners)

alexrj, 2015-12-18:
This is complementary to #763

alexrj, 2015-12-18:
I'll add two new options:
* `--retract-lift-above`
* `--retract-lift-below`

Both will accept an absolute Z. If `below` is zero, lift will take place on the entire print.

alexrj, 2015-12-18:
Done.

closed 2015-12-18 by alexrj