slic3r/Slic3r #3058: Simple OpenSCAD model fails in Slic3r

jbrown123, 2015-09-06:
Environment:
* OpenSCAD version 2015.03-1 
* Slic3r v1.2.9
* Windows 7 Professional SP1 64 bit

I have a very simple model that produces an odd error in Slic3r.  The generated STL looks fine in STLview and in MeshLab (reports no problems).  However, Slic3r hates it.

If I stick two spheres on the end of a cylinder, it fails.  If I hull the same two spheres, it works.  The two spheres are key.  If you just create the cylinder it works fine.

When I view this in slic3r it sees the failing model as being oddly hollow and broken in the 3D tab and non-existent in the preview or layers tabs.  If you load an STL with just the failing object, slic3r says "no layers found" or it just hangs "processing triangulated mesh".

The 3D tab looks like this:
[image]

The preview tab looks like this:
[image]

Here's the offending STL https://www.dropbox.com/s/hlwxxbuoba2q87x/slic3r%20test.stl?dl=0

Here's the OpenSCAD code to generate the above STL:
```
$fn=64;

color("red")
translate([0, -20, 0])
fails();

color("green")
translate([0, 20, 0])
works();

module fails()
{
    translate([-32, 0, 0])
    rotate([0, 90, 0])
    cylinder(r=10, h=64);

    //hull()
    {
        translate([32, 0, 0])
        sphere(r=10);
        translate([-32, 0, 0])
        sphere(r=10);
    }
}

module works()
{
    /*
    translate([-32, 0, 0])
    rotate([0, 90, 0])
    cylinder(r=10, h=64);
    */
    
    hull()
    {
        translate([32, 0, 0])
        sphere(r=10);
        translate([-32, 0, 0])
        sphere(r=10);
    }
}
```

alexrj, 2015-12-19:
Netfabb confirms the model has 18 holes:

[image]

alexrj, 2015-12-19:
You need some overlap between the cylinder and the spheres. It's common practice in OpenSCAD, caused by floating point math.

Slic3r requires a valid model with no holes (https://en.wikipedia.org/wiki/Garbage_in,_garbage_out)

closed 2015-12-19 by alexrj

mrvn, 2015-12-27:
You need to set the same $fn for the spheres and cylinder and pick a value so they align perfectly, as in the vertices of the sphere coincide with the vertices of the cylinder. Depending on directions you have to rotate the spheres or cylinder to get them to align. Overall I would just stick with using hull().