slic3r/Slic3r #3059: [request] extrusionmultiplier for top solid infill

Lord2Vader, 2015-09-09:
Hi,

Printing a lot of watertight thing but with certain filament (petg) it isnt getting watertight. Using a higher multiplier for the whole thing turns it into a stringi smieri mess :)

Would be nice to consider this.

BR vader

Fabian-Multec, 2015-09-09:
I think you can just change the extrusion width for Top solid infill under Print Settings > Advanced > Top Solid infill.

Fabian

lordofhyphens, 2015-09-09:
Fabian, it changes the width, which tells the slicer to put the lines
further apart. The people who want this want the ability to deliberately
overextrude to compensate for some other failing.

If your printer firmware supports the requisite flow rate gcode, I'd
probably look at enabling verbose gcode and then editing the gcode to add a
set to change the flow rate before and after the top solid infill (to
increase it and then reset it). It'd probably be a few minutes' work to
script something up that adds it automatically if desired.

If I find the time, I will see if I can add a "dirty hacks" page to the
advanced to be able to insert these. Will take a while, mostly because I
would have to investigate all of the source to be able to add it.

http://reprap.org/wiki/G-code#M221:_Set_extrude_factor_override_percentage

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Fabian-Multec, 2015-09-10:
Actually, you can find this in the slic3r documentation: 

"Extrusion Width is the thickness of a single filament extruded either in free air or above a surface. It's not the distance of two adjacent paths since some overlap will be generally applied in order to get better bonding."
http://manual.slic3r.org/advanced/flow-math

Lord2Vader, 2015-09-11:
That only means that its not the width between two lines. The distance will be like extrusionwidth-smallamount. But its not that the distance is allways 0.4 at a 0.4 nozzle and its getting thicker. You can see that in the preview. On repetierhost it doesnt take the width to account so if you set a higher extrusionwidth.there will be more.distance between the lines in repetierhost.

Also, why not just add this feature. Iam not willing to change gcode myself. If this would be normal then i could change many things my self which noone does. Its rather common to print such things and it would be a good feature...