slic3r/Slic3r #3060: Change the word "flip" to "mirror"

bcolsen, 2015-09-09:
To start with...this software is great..love it.

My bug:  

I flipped my model along z thinking that is a great short cut for my usual rotate 180 deg along x.
I ended up with the mirror of my model instead of a flip.

This is admittedly a minor change, but 
I just had a "flip" give me left handed threads
on one of my models.

pyrophreek, 2015-09-09:
I have made the same mistake, although I caught it before printing the part.  

Not really a bug, but I agree this would prevent accidents.

bcolsen, 2015-09-09:
It's not a code bug...it's a user interface bug.  A user interface bug happens when ever a users expectation is different from the programs execution. This, unfortunately, becomes very subjective.

As a feature I would really love a real flip the part over button in the tool bar.  I use this more often then rotating the part by 45 deg.

alexrj, 2015-11-04:
I'm not a native English speaker (are you?), but I always thought *flip* and *mirror* had the same meaning. Lots of applications, including SketchUp, use the *flip* word for mirroring...

alexrj, 2015-11-04:
Anyway, I trust you.

bcolsen, 2015-11-04:
Yes I'm a native speaker.  In natural language flip means "to cause to turn and especially to turn over."   A rotation in 3d is implied.   

Yeah sketch-up uses it that way...but that seem to be a language extension of simple image editors.  In a 2d image flipping is the same as mirroring.  You could imagine a photo negative or transparency flipped over would result in a mirrored image.  People who came to computers from film developing where used to this.  

In 3d this is not the case.  If flip a coin images on the coin are not reversed. 
 
Mirroring is more technical of a word than flipping.  Mirroring is well defined in geometry and it's what the program is doing.

I think your users are of the more technical side.  

On the other hand if more and applications use the word flip in this way the users will be retrained and the word may be redefined.  That's the way English works.

I hope this gives you enough information to make a decision.

Thank-you for all you do.

whosawhatsis, 2015-11-04:
As another native english speaker, I'll confirm that "mirror" is a little more clear than "flip" in this context.

alexrj, 2015-11-04:
Thank you for your explanation. I'll change it.

alexrj, 2015-11-04:
Done. I appreciate the pickiness.

closed 2015-11-04 by alexrj

a4jp-com, 2015-11-05:
Nice.