slic3r/Slic3r #3062: no bridge support when it hangs outside the foot print

lahure-c, 2015-09-11:
Hi,

I have set up a example with two bridges, the first bridge has a support but not the second one doest not have support.
Is it because the second one hangs outside the footprint ?

System : Slic3er 1.2.9 and OS Debian Jessie
Files( stl, gcode, ini, png): https://github.com/lahure-c/public/blob/master/20150911

Thank you

[image]
[image]

[image]
[image]

[image]

Mikk36, 2015-09-11:
More likely it seems that the angle is outside of the range that requires support.
To properly test your statement, create a horizontal plane outside the footprint.

lahure-c, 2015-09-11:
> More likely it seems that the angle is outside of the range that requires support.

I tried every angle 
It only works with high values (> 60°), thats is when support is also generated for the oblique part.

But I don't want support for the oblique part, just for bridges.

> To properly test your statement, create a horizontal plane outside the footprint.

In my example there are two bridges, one above the footprint and one outside.
Only the first one gets support. They both have the same horizontal angle
Please see:
https://github.com/lahure-c/public/blob/master/20150911/Bridges.stl

lahure-c, 2015-09-11:
I've made another test
https://github.com/lahure-c/public/blob/master/20150911/Bridges4.stl

This is not a footprint problem
As you can see below, the support is not generated for bridges that start with an edge. But it's generated when it start with a face:

[image]

lahure-c, 2015-09-11:
I made an stl test file with different edge angles or chamfer or filet:
https://github.com/lahure-c/public/blob/master/20150911/Bridges11.stl

Although the edge angle should not change the bridge support decision, it does change it, as if the software was believing that some material is below this bridge edge.

The picture below was got with a support material overhang threshold angle of 22° :
[image]

whosawhatsis, 2015-09-11:
Same issue as https://github.com/alexrj/Slic3r/issues/2222 and others.