slic3r/Slic3r #3063: Customized / manual support placement

Patola, 2015-09-11:
Since slic3r has undergone such a huge change, can the next version provide customized support placement like Simplify3D, CraftUnique Craftware and Autodesk Meshmixer? That would be the silver bullet that would put slic3r ahead of all the other slicers and print hosts.

The way this works in Simplify3D is: you have a dialog box which lets you add support towers using the cursor, remove support towers using the cursor, run the automatic support algorithm to generate the support towers in place according to the parameters set in the configuration and clear all supports in the current part. It colors the supports darker than the actual part so it's easy to distinguish.

If you have customized supports in place but the support option is off, it warns you if you really want this off, lets you select yes or no, then proceeds.

bssstudio, 2016-01-04:
+1

That would be a real killer feature. Many of us would really like to use slic3r instead of anything else as it is opensource and this feature would enable us to do anything with it.

alexrj, 2016-12-18:
@lordofhyphens, aren't we very close to this thanks to your lambda code?

alexrj, 2016-12-18:
Also linking #2606

lordofhyphens, 2016-12-18:
We are closer but the support generation requires a full slice available. I
think that the work done on @bubnikv fork is further but I haven't had time
to evaluate it  in master. I made an early cut/place attempt but needed to
deal with other things.

I do know it uses EdgeGrid.

alexrj, 2016-12-18:
Why does support generation require a full reslice button? Under what circumstances?

lordofhyphens, 2016-12-18:
I was looking at the code,  and its inputs are an array of slices, the code
also makes sure that slicing has finished before continuing. At that point
we don't have much in the way of information about the regions.

Sorry, I meant that the slicing process had to complete (with the current
implementation), not that it had to be restarted.

alexrj, 2016-12-18:
Stupid me, I was thinking this issue was about the reslice button, so discard my comment; totally nonsense. :-D

I was thinking about a very basic approach: create pillars (like we create lambda modifiers) and position them manually, telling Slic3r those are meant to be supports...

lordofhyphens, 2016-12-18:
With a new Ui, we could make that work I think. I also think that it would
be a good thing to properly enable supports as a function of modifier
meshes. A lot of people are often more interested in not putting support
somewhere (although this could be handled by the "support touching build
plate only" option that @bubnikv added

Schleichmichl, 2017-05-17:
I'm sure many people are missing this feature.
In my point of view a first version could contain only the manually deletion of pillars created by Slic3r.
Then the user can decide...

Moving pillars by curser or adding pillars would be even more awesome - but step by step!?

NateTG, 2017-05-17:
A little necromancy here.  I was thinking manual support could be an extension of modifier meshes.  Admittedly the modifier mesh UI isn't the greatest in the world, but it provides facilities for adding and moving stuff.