slic3r/Slic3r #3064: Feature Request: use Cura-Style starting

felicitus, 2015-09-15:
Today I gave slic3r a go, after using Cura for several years.

I noticed that slic3r always seems to do Z-Homing, then moving to the start print position. My printer has clips which slic3r crashed into.

I love the way Cura does start printing, by moving the Z-Axis up to about 30mm and then moving diagonally to the starting position. That way, the extruder doesn't hit the clips.

I already have the following start G-Code:

```
G28 ; home all axes
G1 Z15.0
G1 F200 E3
```

It seems that slic3r always does an additional Z-Homing after the start G-Code, rendering my G-Code useless.

markwal, 2015-09-15:
I don't think so. Slic3r only adds heating if you don't have any in your starting gcode.  Go ahead and generate a gcode file for a small model, paste the whole thing into pastebin or hastebin and maybe a Cura output that you like for comparison and then paste a link back here.  I'm guessing something else is different, like perhaps your origin settings?

justmyopinion, 2015-09-15:
Check your printersettings/Custom G-code/Start G-code and you will see exactly what your G-code file is starting with so where is your "already have the following start G-Code" placed?

felicitus, 2015-09-15:
Here's  the relevant G-Code part of a file I just generated:

```
; generated by Slic3r 1.2.9 on 2015-09-15 at 23:47:37

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.52mm
; infill extrusion width = 0.52mm
; solid infill extrusion width = 0.52mm
; top infill extrusion width = 0.52mm

M107
M190 S100 ; set bed temperature
M104 S210 ; set temperature
G28 ; home all axes
G1 Z15.0
G1 F200 E3
M109 S210 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.200 F7800.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X87.669 Y94.341 F7800.000
```

It uses `G1 Z0.200 F7800.000`, then does `G1 X87.669 Y94.341 F7800.000` next.

In Cura, I believe it would do something like `G1 Z0.200 X87.669 Y94.341 F7800.000`.

My origin is configured as `0,0`.

markwal, 2015-09-15:
You're just asying you want it to start the XY movement before moving the head down?  If so you can add a G1 X87 Y94 (or wherever the center of your bed is) to the end of your start gcode.

felicitus, 2015-09-15:
No, I want XYZ movement at the same time, not after each other.

markwal, 2015-09-15:
OK. Then do a G1 X87 Y94 Z0.2 to the end of your gcode.

felicitus, 2015-09-15:
Can I use variables in the start g-code? Otherwise I would have to adjust the start G-Code for each slice, as start position is different.

markwal, 2015-09-15:
You can even use the start layer height variable like this:
G1 X87 Y94 Z[first_layer_height]

Though it wouldn't matter much because the next motion would adjust it to the proper height anyway.

felicitus, 2015-09-15:
I think the easiest way for now is to simply move XY to the center of the build plate.

felicitus, 2015-09-16:
I have tested this for a few prints now, and this approach gives the issue:

Centering the head in the middle of the platform causes the filament to ooze out onto the print plate.

I'd much prefer Slic3r to take whichever coordinate the printer is set to in the start g-code, then move the head in XYZ direction in one command and not Z, then XY as it's implemented now.

felicitus, 2015-09-16:
That's odd, on another machine slic3r does the right thing: Heating up first, then executing my start G-Code and starts printing. I'll close this again, sorry for the noise.

closed 2015-09-16 by felicitus

lordofhyphens, 2015-09-16:
Your start gcode should include heating instructions, otherwise Slic3r puts
them at the top. I usually park the hotend somewhere it won't ooze, preheat
the hotend to 100C, bring the bed up to temp, then finally bring the hotend
to printing temp and go on with printing.

pyrophreek, 2015-09-16:
The start Gcode is very powerful, and if used properly can do all of the things you want.  To give you an example, here is my start Gcode:
```
G21;                                                           set mm units
G28;                                                           home all axis
M190 S[first_layer_bed_temperature];      begin heating bed
G0 X140 Z5 F400;                                     move to nozzle heating height and position
G90;                                                          set absolute coordinates
G92 E0;                                                     reset extruder distance
M104 S[first_layer_temperature];              set extruder temp and start heating
M109 S[first_layer_temperature];             wait for extruder temp to be reached
G1 X140 Y150 Z0.3 F180.0 E5;               lower nozzle to platform edge height and purge
G1 X140.0 Y140.0 Z0 F3600.0 ;              wipe drool off nozzle on edge of platform
G91;                                                         set relative moves
G1 Z[retract_lift] F1800 E-[retract_length] ;retract and lift for move
G90;                                                         set absolute moves
G1 F3000;                                               Set Feed rate for first move
G92 E0;                                                   reset extrusion distance
```

In my case, my printer head can move past the bed in the Y axis, so I can use the edge of the bed to wipe off the nozzle.  

You should be able to implement something similar to achieve the results you want.