slic3r/Slic3r #3065: Constant Crashes

Chiliyago, 2015-09-16:
Seem lately Slic3r is crashing very regularly and it does so when sitting idle.  Once it has completed the crash there is some type of program that immediatly pops up call plays.tv beta.  Is this secondary program suppose to be part of the distribution now?  If so perhaps it is causing the crash.  I am using 1.2.9 on Windowss 10 64bit machine.  Thanks for any insight you can provide.

lordofhyphens, 2015-09-16:
Your computer is probably infected by malware, my best guess.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

wolfmanjm, 2015-09-16:
A lot of people are reporting crashes in 1.2.9
I'm on Linux so no viruses there :)
Mine usually crashes as soon as I open an STL file, core dumps.
However if I disable background processing it does not crash and I can export gcode file successfully.
This core dump seems to happen on most recent versions I have tried.
1.0.0 version and up just core dump as soon as I run them.
I'm in 14.04 ubuntu Linux.

lordofhyphens, 2015-09-16:
Background processing is one of the first things I disable. Doesn't make
much sense to me at all. I use Linux as well; slic3r 1.2.9 does not crash
for me.

alexrj, 2017-03-18:
We're working a lot on stability.

closed 2017-03-18 by alexrj

foreachthing, 2018-06-14:
If you make changes to infill, support or anything that might re-slice your geometry, DON'T BE IN **LAYERS** TAB! Always switch to another tab (i.e. Preview) _then_ make your changes and wait for it to finish.

lordofhyphens, 2018-06-14:
@babsam11 use 1.3.0, it's released: https://github.com/slic3r/Slic3r/releases/tag/1.3.0