slic3r/Slic3r #3068: Feature request: retract when crossing top/bottom infill layers

mrvn, 2015-09-26:
Slic3r has an option "Only retract when crossing perimeters". I assume that is because a little dribble in the infill won't matter. But activating that option also doesn't retract when crossing top/bottom infill layers. Most important crossing the bottom layer before it gets filled or the top after it was filled. Especially on the first layer this is noticable as the erroneous strand ends up on the outside.

So it would be nice to have more options here: corssing perimeters, crossing shell, top/bottom infill.

alexrj, 2015-12-19:
Can you elaborate more about the problem you want to address?
What's the problem with oozing inside solid infill layers? They're not visible from the outside.

vlast3k, 2015-12-19:
One potential problem this sometimes is that the fillament that has oozed during the move, is missing from the nozzle when it reaches its destination, and then there is underextrusion
yet i cannot remember a specific example right now and i have this option turned on