slic3r/Slic3r #3069: Starting position of loop is often the same

mrvn, 2015-09-26:
I'm printing a box (basically a hollow cube split in half to give box and lid). The simplicity of the box means that there are many identical slices and the slicer picks the same start position for the perimeter (and other loops) for every slice. SInce the starting/ending position is never perfect this leaves a noticable weld/seem on one side of the box going up/down and prefents the lid from fitting on te box in my case. I have to sand off the weld to make it fit.

I tried setting "Seem position: Random" but that doesn't seem to have the desired effect.

PS: Whenn using a thicker shell all the shell loops also start at the same position (as well as the perimeter) since it starts each new loop exactly where the old one ends. It should start the next loop a short distance away from the last to avoid concentrating any error or weak point at the same spot. Maybe that's the same problem.

alexrj, 2015-11-04:
@Vicious-one, can you make a pull request for this? I'd like to keep your name in commit history.

bubnikv, 2016-05-24:
The old C++ line
>   last_pos.rotate(rand() % 2*PI, centroid);
was replaced by
>   last_pos.rotate(fmod((float)rand()/16.0, 2.0*PI), centroid);

The correct solution to get a random angle in <0,2*PI) is
2.0*PI *float(rand())/float(MAX_INT)

Vojtech