slic3r/Slic3r #3070: Defaults for extrusion width of external perimeters for 0.4 nossle are too thin

nikel123, 2015-09-26:
Defaults seems to be 0.4, which is too thin for walls with only 2 shells and the shells don't stick together properly. Extrusion width of 0.5 seems also to be insufficient. 0.6 give me good results. 0.4 seems to be ok, if there more internal perimeters.

ghuser2, 2015-09-27:
My default width is also coming out at 0.4, but it seems to bond well and parts are strong:

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.48mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.48mm
; top infill extrusion width = 0.48mm

nikel123, 2015-09-27:
What is your nozzle size, layer hight, plastic and temperature? My are 0.4 nozzle, 0.2 and 0.1 layer heights, PLA plastic and temperature 220.

kefir-, 2015-09-27:
Slic3r defaults to your nozzle diameter as the perimeter extrusion width, because that's the smallest reliable extrusion width, and that's good to accurately print small details. 

This works fine for me and I think many others. If wider extrusions work better for you, I'd say go for it. Perhaps it's an artifact of your printer. What type of nozzle do you have?

ghuser2, 2015-09-28:
My nozzle is 0.4, with 0.3 layers, opaque black eSun PETG at temps from 225 for small parts to 245 for larger parts.

nikel123, 2015-09-28:
I have my printer only for about a month and it was cheap, i am even really surprised it performs so well. Also PLA i got seems to be pretty viscose at least in comparison to HIPS i got here. As i understand extrusion width = nozzle diameter is pretty much at the edge. Slic3r give me even 0.67 default extrusion width for infill and other perimeters. For some parts 0.4 also works for me, but even not for everything. IMHO defaults should be as foolproof as possible and a thicker extrusion width for external perimeters would only hurt a bit fine details but get much more reliable results with more printers and filaments.

kefir-, 2015-09-28:
I see your point regarding "as foolproof as possible", but in a case like this there's at least one drawback, which is less accurate reproduction of small details on the perimeter. We all have to make some changes from the defaults. 

It could sound like your nozzle is a bit bigger than advertised. Or perhaps there's something else that could cause something like this, but I'd still suggest that you just increase your perimeter width and go with that, if that works better.

curiouspl2, 2015-09-28:
defaults cannot be foolprof because it all depends what you are printing. 
for example structural models (like mechanical gears, pulleys, etc.) get priority of strenght
above aesthetics, while wardrobe, desktop items or sculptures require finest surface details being reproduced , while structural integrity can be maintained by internal perimeter layers inside model, invisible from the outside.

see also problem of steep overhangs, and get familiar with use of modifiers : 
https://github.com/alexrj/Slic3r/issues/2313

also note there are tools like blender which can highlight and select vertices with 3d printing problems, like too steep overhangs or walls being too thin. 
only by use of modifiers you can solve both of these issues if model does have both of problems existing (i.e. with modifier increasing extrusion width for areas with overhangs, while at same time with modifier which decreases extrusion widths for thin wall parts of print)

hoh61, 2015-11-11:
even if this is an old issue, I would like to comment it:
I recently described, that slic3r has a weakness in the flow maths, which generates an error in the material extrusion when changing the extrusion geometry (width, layer height). Too few material will be provided when reducing the extrusion width smaller than nozzle-Diameter+layer height. 
What does it mean for the "defaults": you can optimise your print parameters for one geometry, when changing it, your optimisation is lost. Furthermore, anyone’s parameter couldn't be taken over unless you don't know the geometry behind it.
Therefore the effect of nikel123 is clear assuming an optmised setting evaluated at large extrusion width, while ghuser seems to do his optimisation for small extrusion width. Here I would expect "overextrusion" for larger geometry sizes.
You can use modifiers to smoothen the impact of the maths, or you can improve the maths, making modifiers more or less useless.

lordofhyphens, 2016-07-12:
There isn't really anything here that can be fixed. Closing.

closed 2016-07-12 by lordofhyphens