slic3r/Slic3r #3071: FEATURE REQUEST: Add a "Slice now" button

dcloutier, 2015-09-27:
Using Slic3r on a daily basis, I often have to change many settings for specific print jobs, which trigger the background slicing every time. On larger pieces, it crashes the engine when changing the options and the background slicing has not completed it's run.

My request would be to have a "Slice Now" action button so you can disable background slicing and manually trigger the slicing, see the preview and make adjustments. I know you can export the GCode to trigger the slicing but exporting the GCode is unnecessary for me since I send it the the OctoPrint server (which is a very nice feature btw).

Thanks for the consideration.

justmyopinion, 2015-09-27:
Why not just take a look under Files/Preferences:....

lordofhyphens, 2015-09-28:
Disable background slicing in Files->Preferences. Then it will slice when
you tell the program to "Export Gcode".

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

dcloutier, 2015-09-28:
It is what I am doing already, but having a Slice Now button (which triggers the background process manually for example) would make things much better and consistent. No need to waste time exporting gcode when not necessary. Thanks.

pyrophreek, 2015-09-28:
This is a duplicate of issue #2979.

Agreed this is an important feature to include.

phizz166, 2015-10-05:
+1.

I also have issues with large, complex parts (>50MB STLs) taking a long time to slice, and the background processing not working very smoothly. Before Slic3r had a nice 3D preview, it didn't matter as much that you had to export the code every time you wanted to see the results...but now that you can preview the G-code right in the UI, it's a pain to be exporting over and over again.

My ideal workflow is to load all of the parts that I want to slice, arrange them properly on the bed, set all of the printing parameters, and only then initiate the slicing. After inspecting the preview and confirming that it looks good, I'd export the code separately.

lordofhyphens, 2015-10-05:
Could we not "+1" duplicate issues?

a4jp-com, 2015-10-05:
A +1 button here would be very cool.

lordofhyphens, 2015-10-08:
Working on it now; I think I can factor out the common sections between it and saving gcode. Worst case is that it dumps the gcode to a temp file and then discards it.

lordofhyphens, 2015-10-09:
Keyboard combo to trigger it is ALT-P-N

lordofhyphens, 2015-10-09:
There, pull request submitted. It's in my debian branch as well.

mrvn, 2015-10-16:
Wouldn't it be more useful to have slic3r NOT crash when changing setting while slicing and to make it a little more background then it already is. The GUI sometimes hangs for a while when it starts the slicer in the background. But it's mainly the crashes that are annoying.

alexrj, 2015-11-04:
What about triggering reslice when user activates one of the preview tabs?

lordofhyphens, 2015-11-04:
Well, if you have the code broken out to reslice you could just slave it
into a menu option for the people who really want to cause a reslice for
reasons (like switching presets).

My modification calls the export gcode with a temp filename because I
didn't see how to separate out the slicing from the gcode export.

I'm a little leery of trigging reslice on preview tabs myself; I prefer to
have more control over when slicing and other possibly expensive operations
are done.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

kefir-, 2015-11-04:
I don't really like the idea of reslicing when activating the preview. That's making one action completely dependent on something else, which isn't necessarily what the user wants. 

Couldn't this just be placed in a menu (File -> Reslice, or whatever) and also give it a keyboard shortcut? This won't clutter the gui, and the functionality is always available.

Martronic-SA, 2016-01-12:
+1 for the slice now
(and might be a setting checkbox) to enable/disable automatic slicing.

lordofhyphens, 2016-01-12:
Enable/Disable automatic (background) slicing is in Preferences. Has been
for quite some time.
Slice Now is currently in 1.3.0 in the Plater menu (Alt-P->N)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-09-11:
Please see https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches in about an hour or so, there should be a new download for 'slice-now-background'.

I'm still working on some of the UI-related things (making sure it is disabled if background processing is on, etc).

Tinchus2009, 2016-09-20:
Hi. How wxacty I access the palter menu? Cant find it, ALT P N just opens the menu PLAter but nothing else happens. I have just issued a GIT PULL just to be sure and it says it is uptodate

lordofhyphens, 2016-09-20:
@Tinchus2009 Branch was replaced with the one referenced in 3501; the command is CTRL+S now.

Tinchus2009, 2016-09-20:
I read #3501, but still, ctrl + S does nothing.... something is wrong 
with my instalation I just did GIT PULL again to amke sure and it says 
"updated"...

lordofhyphens, 2016-09-20:
@Tinchus2009 did you pull that branch specifically?
As in:
````
git remote add loh http://github.com/lordofhyphens/slic3r
git fetch loh
git checkout -b background-slice-now --track loh /background-slice-now
````
If you just did `git pull` from your existing checkout, then that is likely pointing at the main master branch for Slic3r (http://github.com/alexrj/Slic3r). 

https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-background-slice-now--a020cd1.zip for a Win32 build of that branch.

There should be a new entry in the File menu for Slice Now (with the keyboard shortcut listed). If that isn't there you have the wrong branch.

To get back to master:
````
git checkout master
````

Tinchus2009, 2016-09-21:
Ok, that is my problem. I though that getting updates from the main 
master branch that would give me the latest devolepment version. My 
intention is, since I print a lot, try  the latests version and 
contribute at least testing, reporting, etc

lordofhyphens, 2016-09-21:
master *is* the main development branch. It's the branch for all accepted features/code. If it's in master on alexrj/Slic3r it is *in* Slic3r and will eventually make it to a release build.

Other feature branches are for developing and experimenting a single feature/fix of Slic3r. They are *proposed* changes/additions/fixes. They won't be in Slic3r until @alexrj accepts them (that's the reason for pull requests), so treat feature branches as extra-experimental ;)

If you want the same effect, you can indeed merge into your master branch, but I can't guarantee things won't get weird for you.

All of my feature branches start as copies of the master branch.

Tinchus2009, 2016-09-21:
Sounds great to me. That way i can help testing those new features too.
A final question: the commands you gave me above will downliad and install that new features not included in the main branch. That cide will be downliaded and merged with the code if main branch? If i go back to main branch all that not included code will be erased again and disapeared ?

lordofhyphens, 2016-09-21:
Switching between master and a branch will generally require a rebuild.

Here is a primer on using git:
https://danielmiessler.com/study/git/

For updating a feature branch with new things from master or another
branch, there is merge and rebase. Read up on what they do and decide which
fits your use case better.

PR #3501 (2016-12-18): Background slice now
Fixes #3071
closed 2016-12-18 by alexrj