slic3r/Slic3r #3072: Feature Request: Ooze Retract While Traveling.  The end of ooze!

Paul--W, 2015-09-28:
Ooze Retract While Traveling: 
<b>A small continuous retraction while moving keeps negative pressure and eliminates oozing.</b>  This is very easy to accomplish: 
- Add an option to Printer Settings, Extruder1, Retraction: Traveling Ooze Retract Length.
This is the amount to retract while traveling.  This length is added back when un-retracting.

I wrote a post processor that I am using to test many techniques, see Feature Request: Coast  https://github.com/alexrj/Slic3r/issues/3014.  I tested Ooze Retract While Traveling with my .571mm bore all-metal hot end with very long (850mm) Bowden tube. It always oozed, and now no more ooze!  I was able to reduce the retract length from 6.8mm to 4.6mm + used 1.7mm of ooze retraction.

A side note: too much retraction with an all-metal hot end can hinder the beginning of the next extrusion (see discussion in issue 3014).

Mikk36, 2015-09-29:
Shouldn't you adjust the amount you retract while moving according to the length of the move?
So you'd retract for example 1 mm for every 100 mm of move?

ghuser2, 2015-09-29:
This is different from just retracting slower, correct?

Mikk36, 2015-09-29:
Yes, normally all of the retraction happens before the move. This feature request asks for additional retraction during the move.

Paul--W, 2015-09-29:
Good question Mikk36,
I will do some further testing.

wfcook, 2015-09-29:
I think you might also want to limit the retraction rate to some maximum in addition to limiting the retractoin distance. If the retraction doesn't get to the maximum distance, it really isn't a problem as long as the negative pressure is kept through the move.

Paul--W, 2015-09-29:
Thanks Bill,
Fortunately when you command a move, with an extrude (+ or - doesn't matter) like:
<pre>G92 E0 ; reset extrusion distance
G1 X100 Y100 E-1.7 F12000  ; move to position 100,100 at 200mms, while retracting 1.7mm</pre>
the move and the extrude both finish at exactly the same time.  The extrude/retract rate is set by the controller to whatever is required.

You will notice that feedrate for the extruder is only stated when it is used by itself for retract/unretract like 
<pre>G92 E0 ; reset extrusion distance
G1 E3 F600  ; extrude 3 mm of filament at 10mm/sec</pre>
And the next move or print command will contain a new feedrate for the following moves (but not for the extruder).

Mikk36, 2015-09-29:
And to put that into layman terms:
X0Y0 -> X100Y100 is 141 mm of travel
Without accounting for acceleration that will take roughly 0.7 seconds to complete.
1.7 mm / 0.7 seconds = 2.4 mm/s, very easy to handle for all extruders.

wfcook, 2015-09-30:
@Paul--W My point is that there are some problems that can occur with very rapid retractions. If you are doing a very short move, say 15mm at 150mm/s, you don't want the full retraction at all. So a maximum retraction speed could be very helpful. I think for pressure control the actual retraction speed may be a more useful number than the retraction distance. I'm postulating that it would be better to set a retract-while-moving rate and a maximum-retraction-distance, so the slicer would calculate the actual retraction distance from the estimated time required for the move, maxing out at some distance to avoid drawing melted filament way up into the heat break.

In other words, for short moves, a shorter retraction distance is required. I suspect the smaller disruption inside the nozzle is beneficial for these short moves.

Mikk36, 2015-09-30:
@wfcook I already suggested for him to test out variable move retraction depending on the move length while having the filament retracted. So for example a 100 mm move would retract 1 mm over that distance while a 10 mm move would retract just 0.1 mm.

Paul--W, 2015-09-30:
You both have valid points.  Testing shows this is more complicated than I realized.
I tested travel over very long distances 250 - 350 mm. At those distances the retract amount I used - 2mm - equates to a very slow retract speed that was not sufficient to control the oozing.  The limiting factor is that I cannot increase the ooze retraction further as it would greatly hinder the extrude at the next segment.  
I ran out of time testing - next time I need to quantify the speed required and maximum distance as you proposed Bill.

DeepSOIC, 2015-10-11:
Hmm, interesting. I was planning to experiment such retract to my postprocessor (SplineTravel). Now I definitely want to experiment.

Since I have a direct extruder, I think the immediate retract may only need to cover the backlash of gears in extruder mechanism, plus a tiny bit more to absorb pressure.

Paul--W, 2015-10-11:
<b>I am withdrawing this feature request.</b>  I will close it tomorrow.
After over a dozen further tests I can say that (for me) the ooze retract while traveling does not help (it may for a different printer).
I used four test distances using two 50mm lines: spaced 200, 150, 128, and 80mm apart like this:

[image]

The real test is to retract the same total length, with the ooze retract on or off.   "Off" (normal retraction) worked the best - less stringing than with a shorter retract + ooze retract.

What I had in fact managed to do with this experiment was to (over time) tune my total retraction amount, and that was what had actually eliminated most of the ooze over these distances.  What one thinks is happening and what is actually happening are not always the same... 

My apologies for taking your time.  

DeepSOIC - please do not let this keep you from experimenting!

closed 2015-10-13 by Paul--W