slic3r/Slic3r #3075: Can't locate OpenGL.pm

pcnate, 2015-09-30:
Can't locate OpenGL.pm in @INC (you may need to install the OpenGL module) (@INC contains:         /home/pcnate/source/Slic3r/lib /usr/lib/perl5/site_perl/5.20.1/x86_64-linux-thread-multi /usr/lib/perl5/site_perl/5.20.1 /usr/lib/perl5/vendor_perl/5.20.1/x86_64-linux-thread-multi /usr/lib/perl5/vendor_perl/5.20.1 /usr/lib/perl5/5.20.1/x86_64-linux-thread-multi /usr/lib/perl5/5.20.1 /usr/lib/perl5/site_perl .) at /home/pcnate/source/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
    BEGIN failed--compilation aborted at /home/pcnate/source/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
    Compilation failed in require at /home/pcnate/source/Slic3r/lib/Slic3r/GUI.pm line 16.
    BEGIN failed--compilation aborted at /home/pcnate/source/Slic3r/lib/Slic3r/GUI.pm line 16.
    Compilation failed in require at (eval 70) line 1.

The error occurs here:

    Use Slic3r::GUI::Plater::2DToolpaths;

I am running OpenSuse 13.2 with everything up to date.

lordofhyphens, 2015-09-30:
Do you have the opengl perl bindings installed?

pcnate, 2015-09-30:
I have not purposely installed anything except that I did encounter issues with Wx. I ended up using CPAN to get that installed.

I will try this later when I get home. Is there anything documented for this?

pcnate, 2015-10-01:
Excuse me while I burn OpenSuse off my drives. I seem to have no idea to get this done on Opensuse.

Might i suggest a test to verify that the OpenGL bindings are installed during setup?

pcnate, 2015-10-01:
Used the following on OpenSuse

    sudo zypper install freeglut-devel
    cpan -i OpenGL

Now this works fine:

    ./slic3r.pl --gui

closed 2015-10-01 by pcnate