slic3r/Slic3r #3077: Material pile up at first layer rim: weird maths

hoh61, 2015-10-02:
As a newbie in 3D printing I was starting to evaluate the proper print settings for my new Vellemann K8400. With some initial settings from the 3D printing community I got a suitable but no good performance. Especially for the first layers I found it strange that the nozzle is rubbing across the material which was deposited before. With the variation of the extrusion speed I could reduce the piled up material at the rim at the cost of a poor filling.
To evaluate this in more depth I tried to set up a test program using the math given in the manual.
To ease my design I generated a plate of 10mmX10mmx0.3mm to check the settings for the first layer.
To get an easy design the infill was orientated in 90°. This design I processed with slic3r 1.2.9 and analysed the resulting GCODE in order to extract some useful structures.

For the first layer I selected a height of 0.3mm and a width of 200%. 
For the infill I found a spacing and material flow which belongs to a width of 0.63mm (210%). Which parameter is affecting this result?
For the skirt the material flow belongs to a width setting of 0,6mm.

The infill is performed like a snake, while every move is depositing the same relative amount of material.
In theory with the long paths the whole area is filled. The small side moves only needs to fill the gap to the skirt.
The skirt itself is so near to the filling, that the gap to the infill is very small, and by the side move the amount of material deposited is five to seven times larger than necessary. I tried this exercise with 0.3mm layer height (seven times too much material) and 0.35mm layer height (5 times too much material).

For this behaviour it is clear that there is lot of excess material at the rim which the nozzle has to push during the next layer.

I tried a quickfix by reducing the deposited material during the sidestep by a factor of 0.2 and the pile-up is removed (generating some failure notifications during the test of the software).
In the next step I will try to provide the data files, the .stl file was refused and all upload now is blocked.

Nevertheless, this is just a quickfix and cannot prevent the careful review of the geometry calculations. 
I processed the same stl file with the cura engine and found, that in this case the infill is done different: the side steps are done without material deposition.

hoh61, 2015-10-02:
please give some hints how I can upload the other files. Github is refusing the GCODE file, the text file and PDF !!!!!

Mikk36, 2015-10-02:
If you measure the actual height of the first layer that you've printed (single traces, not multiple ones bunched up) is the height the same as the first layer height you've set in Slic3r?
Is your filament diameter correct?

hoh61, 2015-10-02:
Sorry, there are some data missing since Github is refusing to accept my files. From the GCODE file you would be able to see that there was only one layer printed (the one which is depicted in the screenshot)
I introduced an Z-Offset of -0.3mm in the printer setting, and for the whole print of the test plate the Z-axis was not moved in the GCODE.
My nozzle width is 0.35mm, filament diameter 1.75 mm. The extrusion length was properly calculated for this thickness (taking the values for the width reported above). I did a fresh adjustment of the table for 0.3 mm distance between print bed and the nozzle; the final thickness of the one layer print was 0.3mm (measured without pile up).
So from mechanical point of view everything should be fine. 
Since I cannot find any other parameter to modify this behaviour I believe in a weakness in the path calculation.

Mikk36, 2015-10-02:
Use pastebin, dropbox, google drive or whatever platform for storing the gcode file and link to that.

hoh61, 2015-10-02:
https://www.mydrive.ch/users/login_guest/483587, pw slic3r.
There you can find the stl-file, the resulting gcode file.
I added the modified GCODE source file too, where I did a modification of the _extrude () procedure.

Mikk36, 2015-10-02:
Unfortunately can't access the files.

hoh61, 2015-10-02:
you have to log in: guest@hoh_ext; slic3r.

Mikk36, 2015-10-02:
As a first step, go to advanced page and set your default extrusion width to 0.35 mm, same as your nozzle width.

hoh61, 2015-10-04:
Setting the default extrusion width didn't change anything. The only difference in the new gcode file belongs to the setting Infill before perimeter, which I disabled in the meantime.
(see testplaettchen_update.gcode).

Mikk36, 2015-10-04:
Oh, also, you shouldn't really try to print 0.3 layers with your 0.35 mm hotend, I'd go for 0.2 or 0.25 at most instead. There's a general rule that for any decent print quality you shouldn't go over 75% of the nozzle width. You'll see a sharp degradation in wall quality if pushing over that.
Also I'm not sure why you're messing with the Z offset instead of zeroing the printer at actual 0 height.

And you didn't answer my previous question about the measured height of actual plastic that has been printed onto the bed.

Could you measure the skirt height after you've printed it?

hoh61, 2015-10-05:
Thank you for your feedback. I will take this for an improvement for the next layers. Actually I'm still dealing with the first layer. The first layer height is limited to the distance between nozzle and print bed. According to the instruction of the supplier this distance should be adjusted to a value of 0.35µm. It can be done smaller, but should never be 0. Having tolerances in the print bed height or non-uniformities of the print bed itself you will damage your nozzle or print bed with this setting. This becomes more serious with two print heads, where you will face small differences even in the z direction between both heads.

Therefore I need the offset correction to compensate the hardware setting, otherwise the height of each printout is too large.

I noticed above the height of the printout, which was 0.3mm in the middle. The test part is waisted already, so I have to redo this part (with the quick fix removed).

hoh61, 2015-10-06:
After a short feedback now a little longer one.

As I mentioned before, I had some problems to get the proper settings for the 3D printer, therefore I thought of a test tool to find the correct settings in a short time. With the mathematic from the manual and some reference code generated for a test plate in slic3r 1.2.9 I sat up a spreadsheet to generate some tst GCODE.
Below there are three examples:
1) Variation of the extrusion temperature (210°C, 200°C, 190°C 180°C bottom to top) and the extrusion factor (0.9, 0.8, 0.7, 0.6 from left to right)
[image]
There you can see my problem with the pile up at the rim. In this case I do not have a perimeter line, but the extrusion is following the same snake like path which is generated in slic3r. I have measured the thickness of the segments in the center and at the rim, the both graphs are showing the results:
Center:
[image]
Rim:
[image]

By the snake-like extruding additional material is supported at the return points of the lines, which will pile up at the rim. Therefore I modified the extrusion in that way that only the long lines are running with material fill (like cura engine). The result can be seen below:
[image]
Once again, the measurement results:
Center:
[image]
Rim:
[image]
Here the height at the rim is slightly reduced as expected, but still much too large.

Since I found, that the area is often not completely filled, I thought about a possible impact of the extrusion width. So I introduced also the possibility to do a variation of the extrusion width. The spacing and flow is calculated according to the slic3r maths, the temperature is set to 200°C as a common temperature, the extrusion width is modified from bottom to top: 0.7, 0.6, 0.5, 0.4 mm, 
The result looks like this:
[image]
Once again, the measurements:
Center:
[image]
Rim:
[image]

with these results I'm complete lost: the deposition thickness is changing with the variation of the width. 

Actually I have no explanation for this behaviour. From these test I see two major issues: 
Why there is the material pile up at the end (possibly hardware relateed)
Why is the thickness changing with extrusion width (wrong extrusion modeling?)

Can anybody give some hints to look after?

Mikk36, 2015-10-06:
Material pile up: I'm guessing you're running a bowden-style extruder? Are you running very low acceleration perhaps? Too slow of a retract?

hoh61, 2015-10-06:
yes, the Vellemann K8400 is a bowden-style extruder. All tests were running at 20mm/s with 80mm/s² as acceleration.
I found some hints, that the extruder speed (?) may be wrong.n  I will work on this topic.

Mikk36, 2015-10-06:
80 mm/s² acceleration is an extremely low acceleration. It's ok for a heavy CNC mill, but not for a printer.
Any half-decent printer should be able to handle 1000 mm/s² and it only goes up from there.

hoh61, 2015-10-08:
Dear Mikk36,
I did a lot of variations with the speed and acceleration and found, that I was out of the range of "printer physics": having these slow accelerations I found heavy material deposition at slow speed since the filament flows too easy. This effect is suppressed at accelerations above 2400 mm/s^2, so I put a default for 2500 mm/s².
Having this setting the results shown above were smoothened but still shows some effects I didn't understood. Especially the area thickness was significantly impacted by the variation of the extrusion factor: 0.9 -> 0.45mm; 0.6-> 0.3mm. 
To determine the correct extrusion factor I followed the calibration procedure from the wiki: I used the _40x10.stl file and got for a nominal extrusion width of 0.4mm a sidewall width of 0.82mm (mean value). This would result in a extrusion factor of 0.49.
I repeated the structure with an extrusion factor of 0.5, the resulting sidewall thickness was 0.45mm, resulting in an extrusion factor of 0.45!!!
With a setting of 0.48 I tried several tests, but a common results was, that the area generated with the mathematics from the manual never achieved a solid plate. An area is deposited by some nearby laying lines, which randomly stick together and can be easily separated. For my chip the surroundings were formed by separate lines, the impression on the top area is, that I may look throught to the lines below.
[image]

Any idea how to continue?

Mikk36, 2015-10-08:
Using a sharpie, a marker or whatever mark 120 mm away from the entry into the extruder onto the filament.
Set the printer up to temperature (200C) and issue G91 (relative moves) and then G0 E100 F100.
This should extrude 100 mm of filament through the extruder.
If it's calibrated correctly you should be able to measure the previous mark 20 mm away from the extruder.
If you're getting some other result, adjust your extruder steps per mm setting until you get exactly the correct amount.

hoh61, 2015-10-10:
Dear Mikk36,

thank you for the tip, I was very surprised, that there was a large error in the setting for the extruder. This is explaining the small extrusion factors mainly.
With the corrected value I did a new calibration (my own procedure)  and found a good setting for the extrusion factor of 0.9. with the proposed procedure I still got reduced extrusion factors, which doesn't make a good surface.
Nevertheless, my problems didn't vanish. I was not able to get a good result, which means, a good surface quality, in any case. Below you may see my concern. I printed two test chips , where I only changed the advanced settings: left: default extrusion width: 0.5mm, first layer : 150%, right: default extrusion width: 0.35mm (nozzle width), first layer: 200%. In each case I had a first layer thickness of 0,35mm and a print layer thickness of 0.2mm 
[image]
On the left chip I got a moderate surface quality, the honeycomb fill is still impressed on the top surface, the surface itself is quite rough. 
In the right chip you have a fine surface structure with a low roughness, which I would rate as a good surface quality.
But every chip has a backside:
[image]
For the left part the first layer is a closed surface, while for the right part there deep groves, giving the impression that the lines doesn't stick together.
As a conlclusion for me I state, that it is impossible to have good print results with variable print settings. The moderate quality of the left chip, the good and bad right chip and the variations in the extrusion factors for single line calibration structures showed, that the scaling of the material uasage is too poor.

This led me to a review of the extrusion model. below your standard model for the calculation of the extrusion rate is repeated:
[image]
This model seems appropriate for all cases, where the extrusion width is larger than the nozzle width + layer height. Unfortunately for all my tests I selected values for w and h, which are outside the limits of this approach. The calculated area for the extrusion  is visualised below:
[image]
I this case the extrusion width is assumed to be smaller than the nozzle width, which seems to be a little bit unrealistic.  In this case the error in the area is depicted by the red segments in the sketch.
As a practical approach a modified model can be applied for print conditions w<n+h: the extrusion below the nozzle is constant, the area contribution is n*h. the excess width is calculated as circle segments with segment width equal to the layer height and segment height equal to half of the excess width. this approach is sketched below. The spacing is calculated by the division of the extrusion area and layer height.
[image]
I modified my test program with this model and found less variations of the deposited material than with the standard model. the layer thickness is still varying a little bit with the extrusion width, therefore it is an improvement, still not a final model.

Since I found it usable on one-layer test structures, I would like to implement it in slic3r for further test.

Is there somebody who can support me , at least with the indications in which part of the source code the calculation of the extrusion width is done (I already found the area calculation)

Mikk36, 2015-10-11:
You shouldn't go over 75% of your nozzle width for the layer height, even for the first layer.

hoh61, 2015-10-11:
I do not understand your argument. Where will be problems with compression? Within the nozzle or outside? 
However, in my understanding there should be no difference as long as you are not flow rate limited or you have the proper amount of material provided. 
I'm quite shure that by my settings I'm not flow rate limited (temperature, speed -> extrusion flow rate), therefore the error seem for me mainly in the material provision.

Mikk36, 2015-10-11:
Outside the nozzle.
But go ahead and try a 20x20x10 box (for example) with .35 or even higher layers and see what the result is and if it's comparable in quality to .25 layers.

hoh61, 2015-10-11:
OK. 
but I would select 0.35 and 0.1 mm to have a quite wide variation of settings.
In the meantime I did a small re-adjustment of the bed distance to 0.3mm.

Mikk36, 2015-10-11:
Could you tell again why you couldn't re-adjust the bed to 0.0mm so you wouldn't have to use the z-offset?

hoh61, 2015-10-11:
because with 0mm the nozzle head is touching the print bead. During moves there is no automatic z-lift for the move, therefore I will scratch the print bed and/or damage the nozzle when running across small irregularities.!
I'hve seen in the Vellemann forum, that thei partially talking from reduced print bed distance in the range of 0.2 or 0.25mm (0.35 in the assembly instructions), but never 0mm!

Mikk36, 2015-10-11:
At 0 mm the head is supposed to just barely touch the bed.
About scratching the bed:
This is the completely default gcode from Slic3r:
; generated by Slic3r 1.2.9 on 2015-10-11 at 11:50:05

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.72mm
; infill extrusion width = 0.72mm
; solid infill extrusion width = 0.72mm
; top infill extrusion width = 0.72mm

M107
M104 S200 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S200 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.350 F7800.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X84.915 Y85.753 F7800.000

I set Slic3r to use -default- profiles for everything and just gave it a 20 mm box.
As you can see, it will home all axis' and then move 0.35 mm away from the bed (first layer height) before doing any XY moves. This means that it will not scratch the bed even on the first layer.
This is also the same behaviour I see from your gcode.
Just remember to keep the z-offset at 0 (the default).

hoh61, 2015-10-11:
Please have in mind:
1) I'm using the K8400,
2) Slic3r is not the only tool for moving the print head.
For slic3r we're on the save side.
But occasionally I will move the print head using the firmware move function.
More appropriate is the commanding using octoprint.
In both cases the printhead is moving directly without review of the z-position.
And in the worst case I don't want to be remembered to lower the print bed by some ugly scratches!

Mikk36, 2015-10-11:
Anyway, how's your result with big layer heights?

hoh61, 2015-10-11:
so, I'm back again with some print results. I printed a box of nominal 10*20*20 mm with the following variations (from left to right:.35mm layer, .35mm width; .35mm layer, .5mm width; .1mm layer, .35mm width; .1mm layer, .5mm width)
Side view:
[image]
and top view:
[image]
So what do I see:
- that for the thicker layer the grooves are larger (what can be expected by physics)
- that for thicker layer there is more "overfill" at the location where the the z-axis is moving
- that for the thicker layer the top filling has a dent (finally not too much surprising, for this configuration the material error is maximal)
- that for the low layer thickness the main groove orientation is changing: for 0.35mm width is along the z-direction, for the .5mm width is within the xy plane.
This is something I didn't expect.

Additionally I measured the samples:
Layer th.   width         height      length      width    dent
0.35            0.35          19.95       9.95        19.85    1.0
0.35            0.5             19.9        9.85        19.65    0.45
0.1              0.35           20.0        9.85        19.7      0.05
0.1              0.5             20.0        9.65        19.5      0.05

A detail I don't understand yet is the fact, that the error in length and width is not a constant but is scaling with the measured value. Could this be a shrinkage effect of the material?

And I'm very surprised that my quick-fix of the excess material gives so good surfaces (see: .1; .35)!
But the surface is still not completely closed: not enough material deposited!

I still ask for some support to update the extrusion model.

Mikk36, 2015-10-11:
I still think there's something off with your calibration though.
Have you tried going through this?
http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide

Specifically this part: "E Steps Fine Tuning"

hoh61, 2015-10-11:
I did something similar. 
I checked the impact on the first layer.
From the mathematics extrusion factor can be taken as fine tuning of the E-steps too. Unfortunately I didn't found a way to change the E-steps during the  printing up to now.
Nevertheless, printing one layer test areas are giving a similar result, in my case I believe a better one:
I did an array of test prints with variation of extrusion factor in one direction and line width in the other.
From the measurement results I can see:
1) an layer thickness proportional to the extrusion factor
2) an increasing layer thickness with decreasing line width.
I observed a thickness variation from 0.25mm (.7mm width) up to .35mm (.4mm width) for the 0.3 nominal layer thickness (at low extrusion factors)
From this result it makes no sense to do a fine tuning in the described way, because if you change the layer height or the extrusion width the finetuning is completely gone.
So from this test I selected the applied extrusion factor of 0.9.
And with the modified model proposed above the thickness variation over linewidth  will be strongly reduced, the layer thickness is constant (within my measurement resolution of 0.05mm).
I will provide a picture later on since I have to re-run this test.

Mikk36, 2015-10-11:
Well, if you read his guide enough, you would have noticed that he basically told to ignore the first 5 or so layers because of their high reliance on the bed leveling.

hoh61, 2015-10-12:
That's why I put some care in proper leveling of the print bed.
Furthermore, the reported values above were deposited on the same positions on the print bed, therefore the error is the same for both conditions. And since the different test areas are located nearby (4 cm distance), any residual leveling error will be negligible.
Furthermore I do not claim a fine tuning but a sufficient adaptation of the hardware variations. the proposed method is for shure more sensitive. From my experience it makes no sense to try a better refinement as long as  I still have the too easy model for the extrusion. 
The hardware calibrations with the extrusion factor also covers the tolerances in the material thickness, since these are factors (or, to be precise for the filament diameter: quotients) in the same formula. So it s only necessary to fine tune one of these settings. Nevertheless, with the hardware values set to the proper values, the correction demand by the extrusion factor will be minimal and the default value of 1 may work sufficiently.
And for different slicers: I  found some usable documentation only for slic3r, therefore I only have the insight of this model, I don't know, what is going on in other tools.
Especially for the repetier host, the preferred tool from Velleman, there are two drawbacks: 1) Vellemann  windows version, 2) I didn't found any documentation about the cura engine.

hoh61, 2015-10-24:
In the meantime I did some further work on hard- and software. I thank Mikk36 for this helpful feedback. My resumee for this topic can be sketched like this:
The initially seen effect was only the peak of the iceberg. 
I cannot trust anything unless I have verified it. This is true even for the supplier of the printer, which sat wrong default values for the calibration.
Everybody is trying to improve the print results, but for the initial settings for the print parameter no reliable procedure is available. What I have seen is dealing only with some facets of this task.
I have experienced that at least for SLIC3R the flow math model is erroneous for most of the print settings I am using. Therefore any fine tuning will be only valid for one configuration, by a small variation the fine tuning is gone.

With this summary I close this issue and will work on the above tasks, issuing separate issues in time.

closed 2015-10-24 by hoh61