slic3r/Slic3r #3078: [feature request] exclude some regions from autospeed

curiouspl2, 2015-10-03:
Autospeed is new cool feature which greatly improves print quality,
unfortunatelly it also slows down prints significantly - only thinner lines of layer are printed up to top speed. 

It would be nice if this could be somehow resolved.
Simplest solution would be to exclude some features, like internal perimeters or infill from autospeed,
but then pressure would build up in the nozzle which would be discharged on the features we would like to have ultra-nice like thin walls or external perimeters.

Threfore bit more complicated hack would be nice, like 'autospeed boost' setting next to each speed setting,  allowing accelerating to predefined 'boost' speed and then deccelerating before move ends. 
ofcourse also acceleration steps/time or some other method to regulate acceleration would be nice. 

This way all 'seams' inbetween print moves will be done at constant pressure like autospeed intends, while parts of model whose visual quality is of second concern will be printed at top speed. 
Also, in case 'boost speed' will be lower than max speed , some features like external perimeters could be printed at lower than maximum allowed speed, while benefitting from having no blobs on start/stop , and with acceleration/decceleration ratios tuned by user to match specifics of extruder/nozzle setup.

thanks for great work and greetings!

lordofhyphens, 2016-05-17:
Something I could think of is to use modifier meshes.

lordofhyphens, 2016-11-27:
You can override the speeds with modifier meshes, which hopefully should be easier to use now. #3590

curiouspl2, 2017-10-23:
Thanks for quick hack.
I may note that using Linear Advance of new Marlin firmware solves most of print quality issues (like blobs) even without autospeed. 
This puts this issue down in priority for some time imho, at least for marlin users with fast enough extruders to be able to enable linear advance (above 700steps/mm extruder Linear advance is not so shabby)

Still with autospeed results are even better :)