slic3r/Slic3r #3080: Detection of thin walls spoils the result.

lonelymyp, 2015-10-03:
1.2.9
gt2 pulley, nozzle 0.3
without function:
[image]

feature enabled:
[image]

appear sharp needles, instead of teeth, that spoil the model.

ghuser2, 2015-10-07:
I have been turning off this function lately, in order to fill-in small areas with more lines.

harriv, 2015-10-20:
Slic3r config and model file would be helpful. Detailed instructions: https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

ghuser2, 2015-10-25:
And actually now, I've been keeping it on. I'm printing some very narrow things (0.6 and 0.3 mm wide) and if I don't turn it on, it will skip these areas.