slic3r/Slic3r #3081: Interface touching the object is not detachable

printerguy, 2015-10-04:
Hello,

I have an issue with the support that cannot be detach if it touches to the object. 

On the other hand, if the interface is under the object there is no problem again. It only occurs if the interface layer is on the actual object.

Examples:

[image]
[image]

How do you solve this problem? My configurations:

```
avoid_crossing_perimeters = 0
bed_shape = 0x0,200x0,200x200,0x200
bed_temperature = 105
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 0.8
bridge_speed = 75
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 6
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0.65
external_perimeter_speed = 20
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.65
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #4AFF6F
filament_diameter = 1.75
fill_angle = 45
fill_density = 20%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 105
first_layer_extrusion_width = 0.65
first_layer_height = 0.4
first_layer_speed = 15
first_layer_temperature = 270
gap_fill_speed = 25
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.65
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 30%
infill_speed = 40
interface_shells = 0
layer_gcode =
layer_height = 0.2
max_fan_speed = 100
max_print_speed = 150
max_volumetric_speed = 0
min_fan_speed = 100
min_print_speed = 10
min_skirt_length = 0
notes =
nozzle_diameter = 0.4
octoprint_apikey =
octoprint_host =
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0.65
perimeter_speed = 40
perimeters = 3
post_process =
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 1
retract_length_toolchange = 1
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = nearest
skirt_distance = 2
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 10
small_perimeter_speed = 10
solid_infill_below_area = 0
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.65
solid_infill_speed = 40
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\nG91\nG1 E10 F300\nG1 X5 Y5 F2000\n
support_material = 1
support_material_angle = 45
support_material_contact_distance = 0.1
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0.45
support_material_interface_extruder = 1
support_material_interface_layers = 2
support_material_interface_spacing = 0
support_material_interface_speed = 40
support_material_pattern = rectilinear
support_material_spacing = 4
support_material_speed = 40
support_material_threshold = 45
temperature = 235
thin_walls = 1
threads = 2
toolchange_gcode =
top_infill_extrusion_width = 0.5
top_solid_infill_speed = 30
top_solid_layers = 3
travel_speed = 150
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0
```

**Test object**: http://www.thingiverse.com/thing:4360

printerguy, 2015-10-05:
Looks like this layers are generated by `generate_bottom_interface_layers` in `Print/SupportMaterial.pm`.   Any idea to make this layers detachable?

lordofhyphens, 2016-06-28:
@printerguy yeah you could splice in the same code in the same perl file that makes the top interface layers, but that function also is currently used for support touching the buildplate. Worth a shot though.

VanessaE, 2019-02-18:
This still needs dealt with :confused: